riscv/riscv-isa-manual #659: Question about C immediate source

yang-le, 2021-06-08:
There is a statement above Table 16.1 that says "Five other immediate bits (5, 4, 3, **1, and 0**) have just two source
instruction bits". Is it correct?

According to my understanding, immediate bit 1 is always sourced from instruction bit 3, and immediate bit 0 is always sourced from instruction bit 2. So I think immediate bits 1 and 0 are the same as bits 17 - 10, that always sourced from the same instruction bit positions.

Correct me if I'm wrong, thank you.

David-Horner, 2021-06-08:
Tables 16.5–16.7 list the RVC instructions is a useful source to look for these C-extension mappings.

Consistent with your analysis, from these tables immediate bits 1 and 0 are explicitly always sourced from instruction bits 3 and 2 respectively.

However, there is the implicit sourcing when these bits are not sourced directly from the instruction bits themselves, but when they are zero and are determined indirectly from the instruction decoding bits.

I see merit in both methods of determining the number of sources.

Yours is simpler and more intuitive, IMHO.

What Andrew chose to highlight is the implementation considerations when creating actual circuitry to effect a translation from RVC to base RVI. 
IMHO, this way be dragons. If the decoding bits are not just determining the location in the instruction for the immediate bits, but rather are also the ones used to determine whether "unsourced" immediate bits are a fixed value, then arguably all the bits of the C instructions are a source for those immediate bits.

it is not surprising you would count the way you did. 
The determination in the RVC section is not expressed in the method used in the base RVI section.
There in Figure 2.4, which "shows the immediates produced by each of the base instruction formats",
one can easily innumerate the number of sourcings, which if you were to include the fixed zeros separately would count five sources for the immediate 0 bit [and 3 sources for bit 1].
If you were to ignore the fixed zero sourcing, then the counts are two for both immediate bits 0 and 1.
I believe many readers would count the way you did, ignoring the fixed zero "source".

aswaterman, 2021-06-08:
I think you'll find David's response helpful.  I'll also add that, while his reverse-engineering of my thinking is accurate, there _is_ cost to the handling of the zero operands.  For wide operands, the cost tends towards zero, but for very narrow operands like these, the control cost is no longer dominated by the datapath cost.

Anyway, since I think your question is mostly answered, and there doesn't seem to me to be anything actionable on the spec side, I'm going to close this issue.

closed 2021-06-08 by aswaterman

yang-le, 2021-06-08:
> However, there is the implicit sourcing when these bits are not sourced directly from the instruction bits themselves, but when they are zero and are determined indirectly from the instruction decoding bits.

Thank you for your reply, but whether to count a fixed zero as a source is not a problem, the problem is if a fixed zero is counted, then it should be counted for all the immediate bits.

For example, immediate bit 3 will have three sources (inst[10], inst[5], and fixed zero) in this situation, other than two. Immediate bit 3 is different from immediate bit 1. Bit 3 has two (or three) sources while bit 1 has only one (or two sources if a fixed zero is counted).

So I think the manual should not categories them into the same group, aka (5, 4, 3, 1, and 0).

aswaterman, 2021-06-08:
I really wouldn't read too far into this.  It's not a normative part of the spec; the goal is only to explain the design pattern.

aswaterman, 2021-06-08:
I'm resolving this issue by deleting the relevant text.  This is not the sort of thing we should be spending time on.

aswaterman, 2021-06-08:
We're getting bogged down here.  The important part is that immediate muxing should be minimized.  The updated text retains the concept without the hyper-focus on detail.

yang-le, 2021-06-08:
OK, I agree. Thank you for the reply.

David-Horner, 2021-06-09:
@kasanovic said I could just ask, so .....
@aswaterman would you please open this issue up so that more discussion can "occur in the open".

Thanks.

aswaterman, 2021-06-09:
You're welcome to reopen it, but I'm not going to address it.

David-Horner, 2021-06-09:
@ANDREW 
i think that's OK because you don't have to spend any more time on it.
There are others who can make the same kind of decisions and weigh in on matters that are a concern to the community.

@kasanovic  
So how do I go about re-opening, as Andrew suggested?
Will you please do this for me?

In this case the  concern to the community is 
1)  readability of the isa manual Vol I and 
2) whether comments are superfluous aggrandizement that should be removed, or pertinent clarificaztion/perspective that should be retained/expanded upon.

Thank you.

David-Horner, 2021-06-10:
@jhauser-us 
We all know how busy krst is.
So, would you please open this issue?

commit f21ac49c (2021-06-08): Delete detailed text surrounding RVC immediates

The metric is vague, and the only important point is the design
pattern, not the exact cost.

Resolves #659