riscv/riscv-isa-manual #660: NTL hints should target caches by their property rather than raw number

jnk0le, 2021-06-10:
As a quick example: L2$ is often an exclusive (bigger ooo cores) and inclusive one(smaller ooo, ino), and sometimes its both within SoCs clustering smaller and bigger cores.

Going deeper:
What if there more than 3 cache levels and the first 3 are all exclusive ones.
Software expecting NTL.L2 to go into inclusive L3 cache is not going to perform as expected.

Does the hint spec dictate the only true cache layout to implement?
Or we are about to see "standardization" of mapping NTL "L*" numbers to actual cache levels?

Instead I'd like to propose skip-levels as:
- innermost exclusive (aka L1, leave any kind of L0 for speculative hardware)
- outermost exclusive (e.g. L2 on big cores, L1 on "little")
- within-group inclusive (e.g. L2 on "little" cores cluster)
- innermost inclusive
- (NUMA/chiplet aware something here???)
- LLC/SLC (goto DRAM)

In case of simpler cache hierarchy some of those can be reduced to the functionally closest one.

aswaterman, 2021-06-10:
> What if there are more than 3 cache levels and the first 3 are all

If you give a mouse a cookie...

jnk0le, 2021-09-18:
ntl by property already implemented.

@aswaterman 
What's the status of ntl for within-group inclusive caches? SMT might mess a bit here with definitions.
If nothing more is added then issue can be closed.

aswaterman, 2021-09-19:
The short answer is that it's up to the implementers to decide what to do: but I expect most will do the obvious thing and treat the inner caches as "private", even though they're technically shared between harts on the core.

closed 2023-02-12 by jnk0le