riscv/riscv-isa-manual #666: mstatus.MPRV behavior in 20190608-Priv-MSU-Ratified 1.11

stanleyleee, 2021-06-17:
Hi, 
     In privilege spec. v1.12 latest draft,  MPRV is cleared when leaving M-mode.

However, if I following the behavior defined in v1.11 ratified version. 

> The MPRV (Modify PRiVilege) bit modifies the privilege level at which loads and stores execute in
all privilege modes.

What if a hart has the following xstatus,  "MRPV = 1, MPP = M, current PRV = U or S"  via executing SRET in Machine mode with "MPP = M, SPP = U or S" ?

The loads and stores shall use the translation and protection mechanisms as Machine mode even if current PRV = U or S?

pdonahue-ventana, 2021-06-18:
Is your question about 1.12 or 1.11?\
In 1.11, it says that MPRV affects "all privilege modes" so that includes U or S.\
In 1.12, MPRV is cleared when leaving M mode.  The behavior of U/S with MPRV=1 doesn't matter since it's impossible for that to happen.

stanleyleee, 2021-06-18:
My question is about v1.11 because I am using the ratified version to implement my design.

However, I have concerns about the behavior of MPRV in v1.11 which might cause U or S mode execute loads and stores use translation and protection as M-mode.

pdonahue-ventana, 2021-06-18:
Yes, this can be a security problem.  I'm pretty sure that's why it was changed.

Technically, I'm not sure if you could ever have U or S execute with M privileges since MPRV uses MPP and the only way to get to U or S is to have MPP!=M and then MRET (and then you can't modify MPP after that).  But you can have U execute with S privileges if MPRV=1 and MPP=S, then you MRET, then SPP=U and you SRET.

pdonahue-ventana, 2021-06-18:
I just thought of a case where you could have U or S have loads/stores execute as if in M mode but you'd have to implement version 0.13 of the debug spec.  Go to debug mode, have the external debugger set MPRV, and then have it resume to S or U.  But debug spec 1.0 does the same thing as priv spec 1.12 and automatically clears MPRV if you resume to a mode other than M so that's no longer a problem.

allenjbaum, 2021-06-18:
Thanks for coming up with that sequence.
I was pretty sure it wasn't impossible, but do we need to worry about it
since you would have needed to have taken over the machine to do it anyway?

pdonahue-ventana, 2021-06-18:
I don't think we need to worry about it architecturally because it has already been fixed in priv 1.12 and debug 1.0.  From an implementation standpoint, if you're using priv 1.11 and/or debug 0.13 then you have to worry about it if you think that buggy M-mode software or buggy external debuggers are a possibility.  Malicious M-mode software or malicious external debuggers might also lead to problems but MPRV is the least of your concerns in that case.

stanleyleee, 2021-06-19:
Thank you for the response.  It seems that M-mode software must use MPRV carefully in priv 1.11.

I thought there might be some sequences to create the scenario without external debugger.

The description for trap return instruction is "An xRET instruction can be executed in privilege mode x or higher".
Therefore,  it shall be legal to execute SRET to exit M-mode and keep MPP value unchanged.

For example, 
1. U-mode trap to S-mode             --> SPP = U
2. S-mode trap to M-mode            --> MPP = S
3. M-mode trap to M-mode again --> MPP = M
4. M-mode execute SRET               -->  PRV = SPP = U,  and keep MPP = M (pop SPP stack only)

This sequence might not be realistic for step 3 and 4, but it's legal behavior in priv 1.11.

allenjbaum, 2021-06-20:
Note that step 3 is basically an irrecoverable error because it overwrites
the Mmode trap state,
 unless the original M-mode handler has saved its trap state, and restored
it before executing the SRET.

zaurbey, 2021-07-08:
#- **@[_tt_]()**

closed 2021-10-31 by gfavor