riscv/riscv-isa-manual #667: A question about the `fence.i` and the `fence r,r` following it

zhengxiaolinX, 2021-06-18:
Hi team,

I am a bit confused about the usage of [`fence.i`](https://github.com/riscv/riscv-isa-manual/commit/03a5e722fc0fe7b94dd0a49f550ff7b41a63f612#diff-b199c03e79944ce7dfb81f721f27d40252590e4daaacc8d9a414a6c4d881f737R967) here, which says to substitute `isb` on ARM, a `fence r,r` is also needed. BTW I cannot find the previous discussion in the memory model task group. I'm wondering what may happen  without the `fence r,r` following and what is the difference between single `fence.i` and `isb`.

Thanks,
Xiaolin

zhengxiaolinX, 2021-06-18:
I have seen the description of [isync](https://github.com/riscv/riscv-isa-manual/blob/master/src/memory.tex#L934) I still cannot get it though -- I wonder if it means that in the software programmer's level, a `fence.i` must be followed by a `fence r,r` in usage?
Thanks in advance.

aswaterman, 2021-06-18:
I'm hoping @daniellustig will chime in here.  My best guess is that `isb` implies additional semantics that often aren't necessary.  There's certainly nothing incorrect about using `fence.i` by itself, even if it isn't quite equivalent to `isb`.

daniellustig, 2021-06-18:
On ARM, a sequence "load; compare; branch; isb; load" implies enforced memory ordering between the two loads, but on RISC-V the sequence "load; compare; branch; fence.i; load" does not.

As Andrew said, there's nothing wrong with using `fence.i` by itself if the intention is to flush the i-cache etc.  OTOH if you are porting from ARM code but have no information about the intention behind the `isb`, then inserting the `fence r,r` ensures the load-load memory ordering in that sequence is also maintained.

zhengxiaolinX, 2021-06-21:
Thanks! I appreciate your help - it really helps a lot.

Thanks,
Xiaolin

closed 2021-06-21 by zhengxiaolinX