riscv/riscv-isa-manual #671: Behaviour when not delegating interrupts

Gallium70, 2021-07-07:
Hi,
Suppose I have a CPU with M and S privilege modes, currently running at M, and some CSRs are set as follows:

```
mstatus.MIE = 1
mstatus.SIE = 1
mie.MSIE = 1
mie.SSIE = 1
mideleg.SSI = 0
```

Now if I set `mip.SSIP = 1` at M-mode, will the CPU jump to M-mode trap handler (mtvec) immediately, or only after executing `mret`, returning to S-mode?

The `mstatus` part in priv spec v1.12 latest draft states that:

> When a hart is executing in privilege mode x, interrupts are globally enabled when xIE=1 and globally disabled when xIE=0.  Interrupts for lower-privilege modes, w<x, are always globally disabled regardless of the setting of any global wIE bit for the lower-privilege mode.

So the SSI should be ignored in M-mode;

While the `mideleg` part says:

> Delegated interrupts result in the interrupt being masked at the delegator privilege level.  For example, if the supervisor timer interrupt (STI) is delegated to S-mode by setting mideleg[5], STIs will not be taken when executing in M-mode.  By contrast, if mideleg[5] is clear, STIs can be taken in any mode and regardless of current mode will transfer control to M-mode.

Does 'can be taken in any mode and ... transfer control to M-mode' mean that it will trap to mtvec? Maybe not immediately since the M-mode is already in control?

aswaterman, 2021-07-07:
In the situation you've constructed, the interrupt will be taken when executing in M-mode.

"So the SSI should be ignored in M-mode" is not right.  Because you haven't delegated SSI to S-mode, SSI is an M-mode interrupt.  The fact that the first "S" in SSI stands for "supervisor" is not relevant here; that's just its name.

Gallium70, 2021-07-07:
> In the situation you've constructed, the interrupt will be taken when executing in M-mode.
>
> "So the SSI should be ignored in M-mode" is not right. Because you haven't delegated SSI to S-mode, SSI is [...]

Thanks for your reply! That solved my question.

closed 2021-07-07 by Gallium70