riscv/riscv-isa-manual #672: Should address-misaligned exception from VU/VS mode set GVA=1?

jameshippisley, 2021-07-08:
The hypervisor spec says: "For any trap (access fault, page fault, or guest-page fault) that writes
a guest virtual address to mtval, GVA is set to 1"

Address-misaligned exceptions can also write a guest virtual address to mtval. It is not clear from this sentence whether they are supposed to set GVA to 1.

Same issue for breakpoint exceptions.

https://github.com/riscv/riscv-isa-manual/blob/c047eaf46d8dccdea10bf75f12f889cd249b5ad4/src/hypervisor.tex#L323-L327

https://github.com/riscv/riscv-isa-manual/blob/c047eaf46d8dccdea10bf75f12f889cd249b5ad4/src/hypervisor.tex#L2006-L2010

gfavor, 2021-07-08:
> The hypervisor spec says: "For any trap (access fault, page fault, or
> guest-page fault) that writes
> a guest virtual address to mtval, GVA is set to 1"
>
> Address-misaligned exceptions can also write a [...]
The mtval definition says "When a breakpoint, address-misaligned,
access-fault, or page-fault exception occurs on an instruction fetch, load,
or store, mtval is written with the faulting virtual address."  I think
this makes clear that the first two cases (which you ask about) are
handled no differently than the other cases listed.

But one can reasonably argue that the parenthetical comment above is
incomplete.  The main part of that sentence says "for any trap that writes
a guest virtual address to mtval", but then provides an incomplete list of
trap causes.  I suggest creating a github issue to this effect.  I've also
cc'ed John to more directly raise this to his attention.

Greg

jameshippisley, 2021-07-08:
Assuming then that the answer is "Yes, these exception types should set GVA=1 when executing in VU/VS mode" I have filed a separate issue requesting the missing types to be added to the listed exceptions (see https://github.com/riscv/riscv-isa-manual/issues/673). So this issue can be closed.

jhauser-us, 2021-07-09:
Greg is right, this is an unintended flaw in the text, repeated for both `hstatus` and `mstatus`.

I don't know that it was really necessary to create another GitHub issue, but since it's done, you can go ahead and close this one, @jameshippisley.  (I don't have the power to do so, but I expect you do.)

aswaterman, 2021-07-09:
Since I'm here, I'll close it.  We can continue the discussion on the other thread.

closed 2021-07-09 by aswaterman