riscv/riscv-isa-manual #673: List of exception types which should set mstatus.GVA is incomplete

jameshippisley, 2021-07-08:
The following two paragraphs provide incomplete lists of the exception types which write a virtual address to mtval/stval. "Address-misaligned" and "breakpoint" should be added to both lists.

(Omitting these exception types from the list leaves room for possible confusion about whether they are supposed to set mstatus.GVA when executing from VS/VU mode, so the omission should be fixed).

https://github.com/riscv/riscv-isa-manual/blob/c047eaf46d8dccdea10bf75f12f889cd249b5ad4/src/hypervisor.tex#L323-L327

https://github.com/riscv/riscv-isa-manual/blob/c047eaf46d8dccdea10bf75f12f889cd249b5ad4/src/hypervisor.tex#L2006-L2010

allenjbaum, 2021-07-09:
mtval is not required to write a virtual address at all, of course, and
which ones it does write are implementation defined.
Platform requirements can mandate that all defined cases will write it,
however, and I 'd expect any platform that supports hypervisors will
mandate it.

JamesKenneyImperas, 2021-07-09:
I think it would also help to be more explicit about what qualifies as a _guest virtual address_ here. Clearly an access using `vsatp` translation counts, but does an access when `vsatp` translation is _disabled_ but `hgatp` translation _enabled_ set GVA=1 or not? The input address in this case is a guest physical address.

gfavor, 2021-07-09:
On Fri, Jul 9, 2021 at 2:58 AM JamesKenneyImperas ***@***.***>
wrote:

> I think it would also help to be more explicit about what qualifies as a *guest
> virtual address* here. Clearly an access using vsatp translation counts,
> but does an access when vsatp translation is [...]
The GVA is the VA as calculated by guest load/store instructions -
irrespective of whether the GVA is explicitly translated to a GPA (due to
vsatp.MODE!=Bare), or trivially translated to a GPA (due to
vsatp.MODE==Bare).

Greg

JamesKenneyImperas, 2021-07-09:
So you are saying that GVA should be set to 1 for _any_ load/store instruction executing in VU/VS mode (and presumably by any HLx/HSx instruction also) irrespective of `vsatp` or `hgatp` state - correct?

I think it is worth stating this explicitly.

gfavor, 2021-07-09:
What I'm saying is that GVAs when operating in VS/VU-modes are exactly the
same as VAs when operating in S/HS/U modes, i.e. they are the addresses
calculated by the load/store instructions - irrespective of what happens to
those addresses after that.

Greg

jhauser-us, 2021-07-18:
Addressed by pull request #678.

gfavor, 2021-07-18:
John, shouldn't Breakpoint fault also be added to the parenthetical "for any trap" list?

jhauser-us, 2021-07-18:
> John, shouldn't Breakpoint fault also be added to the parenthetical "for any trap" list?

Right.  Fixed now, hopefully.

aswaterman, 2021-07-18:
Resolved by #678

closed 2021-07-18 by aswaterman