riscv/riscv-isa-manual #682: HLVX.HU/W behavior for readonly pages does not match spike

jameshippisley, 2021-07-21:
According to the following paragraph, HLVX.HU/WU instructions will give a page fault when accessing a page with R set and X not set.

https://github.com/riscv/riscv-isa-manual/blob/e376cfc42438dab01c60e62222d450dd423751da/src/hypervisor.tex#L1726-L1730

In spike however, HLVX.HU/WU are implemented as analogous to loads with mstatus.MXR=1, i.e. they can read a readonly page and an execute-only page.

The statement "execute permission takes the place of read permission during the translation" might leave ambiguity about which parts of the translation process are affected - i.e. should a Load or an Instruction fault be taken in the case of a page fault, access fault, misaligned exception. In the spike implementation it is a Load fault.

Assuming that the implemented spike behavior is what is intended (which I believe makes sense), I suggest changing the language to match the MXR language:

"The HLVX.HU/WU instructions are equivalent to the HLV.HU/WU instructions but modify the privilege with which loads access virtual memory. For HLV.HU/WU, only loads from pages marked readable (R=1 in Figure 4.17) will succeed. For HLVX.HU/WU, loads from pages marked either readable or executable (R=1 or X=1) will succeed."

aswaterman, 2021-07-21:
The first sentence doesn't strike me as ambiguous; it suggests that read permissions do not factor into the permission check at all.  If so, that would suggest Spike is incorrect.

@jhauser-us Is there a good reason to define this HLVX to require X, rather than R || X, as would be the case for the MXR feature?  The normal use case is to emulate an instruction that was successfully fetched but raised an illegal/virtual instruction exception--meaning we know the page already had execute permissions.  So either definition would appear to suffice.

jhauser-us, 2021-07-21:
> The normal use case is to emulate an instruction that was successfully fetched but raised an illegal/virtual instruction exception.

There are concerns that under certain unusual conditions it may be necessary for a hypervisor to emulate one or more instructions that did not just trap from an illegal/virtual instruction exception.  For example, not long ago the privileged ISA mailing list had discussions about a situation where a hypervisor would need to sequentially emulate *all* instructions executed by a virtual machine while a certain subfield of an emulated CSR had a particular value.  We modified the hypervisor spec to avoid that specific case, but I don't believe anyone can definitely prove that a similar one can never arise again.

HLVX requires execute permission to prevent any chance of a hypervisor being tricked into emulating execution of memory bytes to which the guest has only read/write permission.

HLVX is unambiguously a load instruction, so if its explicit memory read raises an exception, it will be one of the load exceptions:  load address misaligned, load access fault, load page fault, or load guest-page fault.  An instruction fetch trap would occur for HLVX only during the instruction fetch of the HLVX, same as for any other instruction.

aswaterman, 2021-07-21:
Thanks for that explanation; it checks out to me.  I'll make a PR to fix Spike.

aswaterman, 2021-07-22:
Closing, as this is apparently not an ISA spec issue.

closed 2021-07-22 by aswaterman

JamesKenneyImperas, 2022-01-07:
@aswaterman, could you please add John Hauser's statement quoted below as a non-normative note, perhaps in section 8.3.1? This isn't obvious and I can't see it explicitly stated anywhere. Thanks.

> HLVX is unambiguously a load instruction, so if its explicit memory read raises an exception, it will be one of the load exceptions: load address misaligned, load access fault, load page fault, or load guest-page fault.

aswaterman, 2022-01-07:
I disagree that this deserves clarification, but it's getting to the point that acquiescing takes less time than arguing against.

JamesKenneyImperas, 2022-01-07:
Ok, but I've been asked this question twice from different sources now, so I think some people are still confused.

pdonahue-ventana, 2022-01-07:
I also talked to someone about a year ago who thought that a page fault on the load portion of HLVX would be reported as an instruction page fault.  Not because the spec says so but because that was the intuitive behavior to that person (it's "a load that emulates an instruction fetch" rather than "a load which requires execute permission rather than read permission" which is what the spec actually says).  So adding a sentence can help people who think that way.

aswaterman, 2022-01-10:
Okay, much to my surprise, this has come up often enough that it's worth clarifying.

aswaterman, 2022-01-10:
77a05455a0b51ba1c2796a8a455f3e3bc3041621