riscv/riscv-isa-manual #686: Typo in LUI U-immediate

stnolting, 2021-07-22:
I think there is a typo here:

https://github.com/riscv/riscv-isa-manual/blob/03ccc08a560f1a0e026eb306b2dbce4f133cc7fe/src/rv32.tex#L575-L578

Shouldn't this be
```
 LUI places the **20-bit** U-immediate value into 
 the destination register {\em rd}, filling in the lowest 12 
 bits with zeros. 
```
as the U-immediate from the instruction word is only 20 bit wide (20-bit immediate + 12 LSB-aligned zeros = 32-bit)?

nick-knight, 2021-07-22:
I guess it's a question of, What is the "immediate": is it a length-32 bitstring, whose lowest 12 bits are zeros, or is it just the leading 20 bits of that bitstring? From my (software) perspective, the immediate's _value_ is the former, while its _encoding_ is the latter. Since the spec uses the word "value", I accept it as written.

stnolting, 2021-07-22:
> For what it's worth, here is how I reword this sentence in my head:
>> LUI places a 32-bit value in the destination register {\em rd}, taking the upper 20 bits from the U-immediate field and zeroing the lower 12 bits.

Sounds good!

nick-knight, 2021-07-22:
To be clear, I am not proposing a change to the text, just suggesting its intended interpretation. The community strongly opposes churn in normative, ratified text in the base ISA, unless there is a clear defect. Since I see no reasonable alternative interpretations of this sentence, I don't see a strong argument that it is defective.

kasanovic, 2021-07-22:
U immediates are explained explicitly in earlier section 2.3 on immediate encoding variants:
"Similarly, the only difference between the U and J formats is that the 20-bit immediate is shifted left by 12 bits to form U immediates and by 1 bit to form J immediates."
There is some redundancy in the LUI description, but I don't think there's actually any conflict in descriptions.

stnolting, 2021-07-22:
You have convinced me :wink:
Thanks for clearing!

closed 2021-07-22 by stnolting