riscv/riscv-isa-manual #702: SIP view in M-mode when an interrupt is delegated

BiwefC, 2021-08-06:
The privileged v1.11 says:

> Restricted views of the mip and mie registers appear as the sip/sie, and uip/uie registers in S-mode and U-mode respectively. If an interrupt is delegated to privilege mode x by setting a bit in the mideleg register, it becomes visible in the x ip register and is maskable using the x ie register. Otherwise, the corresponding bits in x ip and x ie appear to be hardwired to zero.

And in v1.12 draft:

> Restricted views of the mip and mie registers appear as the sip and sie registers for supervisor level. If an interrupt is delegated to S-mode by setting a bit in the mideleg register, it becomes visible in the sip register and is maskable using the sie register. Otherwise, the corresponding bits in sip and sie appear to be hardwired to zero.

So does it mean that if an interrupt is not delegated to S-mode, it is not visible in SIP even in M privilege mode?

allenjbaum, 2021-08-06:
That was always my interpretation.
 If you need to look at IP bits in Mmode, read MIP, not SIP.
Reading SIP gives you the view as if it were being read by S-mode (if
you're allowed to read it at all)

aswaterman, 2021-08-07:
That's right.

As a general principle, we try to avoid giving instructions/CSRs privilege-mode-dependent behavior, because that is a recipe for virtualization holes.  The behavior should either be the same across privilege modes or should trap in less-privileged modes.

In this case, had we chosen to define sie/sip to behave differently between M and S modes, then we would've erected a roadblock to classically virtualizing M-mode by running it in S-mode.

closed 2021-08-09 by BiwefC