riscv/riscv-isa-manual #721: Question about exception handling for misaligned store instruction

taoj33, 2021-08-25:
Hi,
   Based on  RVWMO model, misaligned store instruction may be decomposed into multiple memory operations and may not be guaranteed to execute atomically.
   My question is about hardware exception handling when those memory operations may behave differently in terms of exception.
   For example, if hardware decompose the misaligned store into two memory store operation and only the second store will trigger exception. 
   So is it allowed to commit the first store operation when the second store triggers exception？Since it is much easier for hardware implementation.
   
 Thx

jiamenghan, 2021-09-07:
My opinion is that the unaligned first store execution is allowed in this case.
Is it right?

aswaterman, 2021-09-07:
That's right: since the access can be decomposed into a sequence of smaller accesses, it follows that the sequence might partially complete before an exception is detected.

closed 2021-09-07 by aswaterman

taoj33, 2021-09-07:
> That's right: since the access can be decomposed into a sequence of smaller accesses, it follows that the sequence might partially complete before an exception is detected.

Thanks for reply.  
I am also worried about a situation related to this implementation. What if software check store data result in the exception handler or skip the replay of store instruction after the exception handler.  The partial store result may not make sense to software programmer since misaligned instruction do not have vstart value like vector instruction indicating partial commit result.  
I am not sure if it is an issue for such implementation.

aswaterman, 2021-09-07:
It's a fair question, but it's OK in practice because:
- The effect is not visible within a single application thread because stores are idempotent: completely reexecuting the store results in the same final state after the store completes.
- The effect _could_ be visible to another thread, but the ISA says that misaligned accesses are not guaranteed to be atomic.  The partially completed store has similar effects in practice to a non-atomic store.  Furthermore, because these accesses aren't atomic, they shouldn't be used for inter-thread synchronization, anyway.

Note this property is not unique to RISC-V (e.g., IIRC, SVE doesn't have the equivalent of RVV's vstart).

daniellustig, 2021-09-09:
I agree with the non-atomicity part but will quibble with the re-execution part.  If two harts do two stores to the same misaligned address, and the stores partially fault in such a way that part of both gets re-executed, then a third hart observing the value of that address could see the memory location (at least partially) alternating between the values written the two (repeating) stores.    I agree that shouldn't be a problem for something like C++ with DRF semantics, but it technically wouldn't be allowed under RVWMO as currently defined.  RVWMO would expect each component of each misaligned store to happen only once, not to appear as if it may have happened multiple times.

aswaterman, 2021-09-09:
@daniellustig Indeed, the "in practice" qualifier I used was doing a lot of heavy lifting... they are definitely not equivalent.  The upshot is that misaligned accesses just shouldn't be used for synchronization.

allenjbaum, 2021-09-10:
You might want to say that explicitly - too many odd corner cases, can't
guarantee, etc. Intel finally went that route, I believe.

aswaterman, 2021-09-10:
This spec does remark that misaligned loads and stores "require additional synchronization to ensure atomicity"