riscv/riscv-isa-manual #729: "D Standard Extension for Double-Precision Floating-Point" is rather incomplete

soc, 2021-09-01:
The first chapter specifies ...

> The D extension depends on the base single-precision instruction subset F.

... which is insufficient where the F extension is specific to its bit-length (compared to talking about the _fcsr_), e. g. ...

> For single-precision floating-point, this corresponds to the pattern 0x7fc00000.

... where the equivalent information is missing for D.

nick-knight, 2021-09-01:
I'm not sure I understand. Is your concern that the F-extension says:
> The canonical NaN has a positive sign and all significand bits clear except the MSB, a.k.a.
the quiet bit. For single-precision floating-point, this corresponds to the pattern 0x7fc00000.

while the D-extension doesn't spell out the pattern? It's fairly straightforward to derive it:
`0 11111111111 1000000000000000000000000000000000000000000000000000 = 7FF8 0000 0000 0000`

I'm not convinced this particular omission is a defect. A correct reading of the spec is sequential, meaning that definitions (like "canonical NaN") in earlier chapters remain in force in subsequent chapters.

soc, 2021-09-01:
Ok.

closed 2021-09-01 by soc