riscv/riscv-isa-manual #740: about the immediate

runninglinuxkernel, 2021-09-16:
here is two instruction:
```
	addi a1, t0, 0x800
	addi a1, t0, 0xfffffffffffff800
```
here is the compile result:
```
src/asm_test.S: Assembler messages:
src/asm_test.S:12: Error: illegal operands `addi a1,t0,0x800'
```

in the spec, it said the ADDI instruction using the sign-extended 12-bit immediate , like Bit[11:0], why i am using 0x800 is wrong operands, the Bit 11 is 1. is it compiler's issue?

aswaterman, 2021-09-16:
The valid immediate range for ADDI is [-2048, 2047].  0x800 is 2048, which is too large.  0xfffffffffffff800 works because it corresponds to -2048.

runninglinuxkernel, 2021-09-16:
why the compiler donot know 0x800 is valid for Bit[11:0] in instruction's format?

aswaterman, 2021-09-16:
Because 0x800 is _not_ valid.  It represents a number outside of the valid range [-2048, 2047].

nick-knight, 2021-09-16:
This is a common confusion with the RISC-V assembly language and its implementation in the GNU assembler.

It is true that `0x800` is a valid 12-bit `imm` field in an I-type instruction. It corresponds to the immediate value `-2048`. But the GNU assembler parses `0x800` as the unsigned integer 2048 --- you need an explicit minus sign, `-0x800`, if you want to write `-2048` [[GNU assembler manual](https://sourceware.org/binutils/docs/as/Integers.html#Integers)]. 

The GNU assembler also has this nifty feature that you've discovered, which allows the programmer to provide the XLEN-bit sign-extended immediate as its two's complement representation reinterpreted as an unsigned integer. For example, on RV32I, you can encode the immediate `-2048` as `0xfffff800` or `4294965248` (binary and octal versions too). This does not work on RV64I: there you can instead encode `-2048` as `0xfffffffffffff800` or `18446744073709549568`.

This is all specific to `addi`. For example, `lui` and `c.lui` have different, exciting behaviors, which I attempted to start documenting [here](https://github.com/riscv-non-isa/riscv-asm-manual/pull/66). By the way, these questions about the assembly language really belong over at [riscv-asm-manual](https://github.com/riscv-non-isa/riscv-asm-manual), and your contributions are welcome there.

runninglinuxkernel, 2021-09-17:
It make sense.  Thanks @nick-knight @aswaterman

closed 2021-09-17 by aswaterman