riscv/riscv-isa-manual #741: Presence of SCOUNTEREN CSR in the absence of S mode

aefimov-cb, 2021-09-21:
Hello,

I wish to clarify presence of SCOUNTEREN CSR in the absence of S mode (in M mode only or M+U modes configurations).

For MCOUNTEREN CSR Privilege ISA spec states (https://github.com/riscv/riscv-isa-manual/blob/master/src/machine.tex, line 1788):
```
In systems with U-mode, the {\tt mcounteren} must be implemented, but all
fields are \warl\ and may be hardwired to zero...
In systems without U-mode, the {\tt mcounteren} register should not exist.
```
It seems logical.

But for SCOUNTEREN the spec simply states (https://github.com/riscv/riscv-isa-manual/blob/master/src/supervisor.tex, line 570):
```
{\tt scounteren} must be implemented. 
```
One might think that this CSR should be implemented in the absence of S mode. It doesn't seem logical for me.

Please, clarify.

Regards,
Andrey

aswaterman, 2021-09-21:
`scounteren` is defined in the S-mode chapter.  Implicitly, that means that any mandates within the chapter apply only if S-mode is implemented.  Put another way, if S-mode is not implemented, the entire chapter can be ignored.

aefimov-cb, 2021-09-21:
Thank you very much for the quick reply.

closed 2021-09-21 by aefimov-cb

aswaterman, 2021-09-21:
Glad to help!