riscv/riscv-isa-manual #743: Clarity for condition when virtual instruction trap will occur

saadkhalid-10xe, 2021-09-23:
In privileged spec (Hypervisor Extension, Version 1.0.0-rc), it is mentioned:

> When V=1, a virtual instruction exception (code 22) is normally raised instead of an illegal instruc-
tion exception if the attempted instruction is HS-qualified but is prevented from executing when
V=1 due to insufficient privilege or because the instruction is expressly disabled by a supervisor
or hypervisor CSR such as scounteren or hcounteren

The condition for scenarios where virtual exception will take place of illegal instruction exception is a bit ambiguous. Its mentioned that attempted instruction should be HS-qualified and,
1) "insufficient privilege" 
or
2) "because the instruction is expressly disabled"

The other way of reading it is,
1) HS-qualified and "insufficient privilege" 
or
2) "because the instruction is expressly disabled"

Do you think it should have more clarity?

allenjbaum, 2021-09-23:
Are yo asking if is this whether the definition is interpreted as A & (B or
C) or as (A&B) or C ? (A=HS qualified, B= insufficiaent priv, C=expressly
disabled)

saadkhalid-10xe, 2021-09-23:
> Are yo asking if is this whether the definition is interpreted as A & (B or C) or as (A&B) or C ? (A=HS qualified, B= insufficiaent priv, C=expressly disabled)

yes.

jhauser-us, 2021-11-12:
Addressed by #773.

closed 2021-11-12 by jhauser-us