riscv/riscv-isa-manual #744: mseccfg values when Smepmp and Zkr aren't implemented

pdonahue-ventana, 2021-09-24:
mstatus.SBE is defined to be a WARL bit that is hardwired to 0 if S-mode is not supported.  The mseccfg CSR defines 5 bits but doesn't have similar language.  Technically, I think that this requires all implementations (regardless of Zkr/Smepmp support) to follow the definitions supplied in Zkr and Smepmp.  I'm sure that's not the intention.

It looks like \*envcfg may need something similar for the Sstc and Zicbo\* bits.

aswaterman, 2021-09-24:
This detail will be addressed when those specs are integrated back into the mainline.  Of course, as you say, the expectation is that the bits will be hardwired if the corresponding extension is not implemented.

aswaterman, 2025-12-09:
Although there's no functional problem with the back-references, I think it would be best for the definition of these bits to be integrated into the section where `mseccfg` is defined.