riscv/riscv-isa-manual #745: Question about RV32I Base Instruction Set List

xingmingjie, 2021-09-30:
In Chapter 25 RV32/64G Instruction Set Listings, FENCE.TSO and PAUSE are listed in RV32I Base Instruction Set. PAUSE is also described in Chapter 4 “Zihintpause” Pause Hint, Version 2.0. Does this mean that Zihintpause belongs to RV32I base set?

Section 2.7 says that FENCE.TSO is an optional instruction.  Does this mean that RV32I base set may not contain it?

aswaterman, 2021-09-30:
Section 2.7's use of the word "optional" is unfortunate.  _All_ RISC-V implementations implement FENCE.TSO.  It was encoded using unused bits in the instruction, which old implementation were required to ignore.  Implementations that ignore bit 31 and simply executed the instruction the same as FENCE RW,RW are correct, because FENCE RW,RW is strictly stronger than FENCE.TSO.

I will improve the description in the spec.

The story with PAUSE is related but separate.  Because PAUSE is a HINT, _all_ RISC-V implementations can execute PAUSE.  Implementations that don't implement Zihintpause will simply execute it as a NOP (which is functionally correct).  Implementations that do implement Zihintpause will briefly slow down the processor (which is functionally the same as a NOP).

xingmingjie, 2021-09-30:
Other problem is that Chapter 2 says RV32I contains 40 unique instructions, but the RV32I Base Instruction Set list actually have 42. it's a little confusing.

aswaterman, 2021-09-30:
I fixed the FENCE.TSO description in e511c4664368355e6d85b8e487fdfdc5d7de44e4

FENCE.TSO and PAUSE are subsets of the FENCE instruction, so the remark that RV32I contains 40 unique instructions is correct, though admittedly confusing.  We might revisit that description in the future.

xingmingjie, 2021-09-30:
Great! Thanks.

closed 2021-09-30 by xingmingjie

aswaterman, 2021-09-30:
Thank you for helping to improve the spec!