riscv/riscv-isa-manual #752: One question about mstatus.SPP

unicornx, 2021-10-12:
Hello,
I'm reading Volume II: Privileged Architecture, Document Version 20190608-Priv-MSU-Ratified.
I have one question about  3.1.6 Machine Status Register (mstatus), I doubt why we define a field SPP for mstatus and how should we use this bit?
I also notice the spec mentioned that `To support nested traps, each privilege mode x has a two-level stack of interrupt-enable bits and privilege modes.` I wonder if mstatus.SPP is used for nested traps case, but I still don't know how to use it when nested traps happens.
Who can help point it out, thanks in advanced.

unicornx, 2022-07-26:
Who can help reply this? Thanks.

gfavor, 2022-07-26:
The following sections in the Priv spec describe the respective SPP bits:
- 3.1.6  mstatus
- 4.1.1 sstatus
- 8.2.1 hstatus

Taking mstatus.SPP as an example, the spec says the following:
_To support nested traps, each privilege mode x that can respond to interrupts has a two-level
stack of interrupt-enable bits and privilege modes. xPIE holds the value of the interrupt-enable bit
active prior to the trap, and xPP holds the previous privilege mode. The xPP fields can only hold
privilege modes up to x, so MPP is two bits wide and SPP is one bit wide. When a trap is taken
from privilege mode y into privilege mode x, xPIE is set to the value of x IE; x IE is set to 0; and
xPP is set to y._

*RET instructions, as one would imagine, "pop" this two-level stack.

unicornx, 2022-07-27:
> nested traps

Thank  you gfavor. But it seems still does not resolve my problem :(.

Let me clarify again. To my understanding,  for example, when  the hart is in machine mode, and when it wants to return from a trap, we can call MRET. According to spec, my understanding is hart will check mstatus.MPP to know which mode to return back to.  So I wonder why we define mstatus.SPP, it looks useless.

At the same time I notice in spec, when it talks about mstatus.xPP, the context is about `nested traps`. So I guess mstatus.SPP may have relationship with this, but I have not image how it works and when the hart will check mstatus.SPP.

Hope this upon clarifies my question, thanks in adv.

aswaterman, 2022-07-27:
mstatus.SPP is a shadow of sstatus.SPP (the latter of which is obviously necessary). Including this field in mstatus thus adds no additional architectural state, but it preserves the useful property that mstatus is a superset of sstatus. This accelerates M-mode context switches, for example.

unicornx, 2022-07-29:
> mstatus.SPP is a shadow of sstatus.SPP (the latter of which is obviously necessary). Including this field in mstatus thus adds no additional architectural state, but it preserves the useful property that mstatus is a superset of sstatus. This accelerates M-mode context switches, for example.

Thanks, I will close this issue now.

closed 2022-07-29 by unicornx