riscv/riscv-isa-manual #761: Plans to define "Zipmp" extension?

stnolting, 2021-10-29:
In the latest release of the specs (https://github.com/riscv/riscv-isa-manual/releases/tag/draft-20211029-4e4dfff) the `Zicntr` (CPU counters) and `Zihpm` (hardware performance monitors) extensions were defined.

I am curious if there are any plans to define something like a `Zipmp` extension that says something about the _physical memory protection_ being implemented or not?!

aswaterman, 2021-10-30:
Assigning names to the various widgets in the privileged architecture is indeed a topic of discussion within RVI, but there isn't a concrete plan yet.

FWIW, the name for PMP would be something like `Smpmp`, since `Sm` is the prefix we're adopting for extensions visible only to M-mode.  Debug extensions that aren't visible to M-mode would be prefixed differently still.

stnolting, 2021-10-30:
Thanks for clearing!

One more question:

> since Sm is the prefix we're adopting for extensions visible only to M-mode

Why the "`S`"? Doesn't that mean it is something related to `S`upervisor mode?

aswaterman, 2021-10-30:
It would be more natural to use the “M” prefix here, but since “M” already means “multiply” in this namespace, we’ve instead chosen “Sm” as the prefix for M-mode extensions.

closed 2021-10-31 by stnolting