riscv/riscv-isa-manual #762: Clarity for FIOM in menvcfg, senvcfg, henvcfg

BiwefC, 2021-11-02:
In privileged v1.12 public review:
> If bit FIOM (Fence of I/O implies Memory) is set to one in menvcfg, FENCE instructions executed in modes less privileged than M are modified so the requirement to order accesses to device I/O implies also the requirement to order main memory accesses. Table 3.8 details the modified interpretation of FENCE instruction bits PI, PO, SI, and SO for modes less privileged than M when FIOM=1.

For the implementation that the FENCE instructions ordered accesses to device IO are always ordered as though they also access memory, can we just hardwire FIOM to 1'b1? Or just let the FIOM be read/write but the value of it will not affect the hardware behavior?

gfavor, 2021-11-02:
On Mon, Nov 1, 2021 at 8:20 PM BiwefC ***@***.***> wrote:

> In privileged v1.12 public review:
>
> If bit FIOM (Fence of I/O implies Memory) is set to one in menvcfg, FENCE
> instructions executed in modes less privileged than M are modified so the
> requirement to [...]
Since FIOM is not a WARL bit, it cannot be hardwired.  But an
implementation that doesn't distinguish between "memory" and "device
I/O" accesses on FENCE instructions (or at least treats "device I/O"
accesses as both "device I/O" and "memory" for FENCE purposes) can simply
ignore FIOM.

Greg

BiwefC, 2021-11-02:
> On Mon, Nov 1, 2021 at 8:20 PM BiwefC ***@***.***> wrote: In privileged v1.12 public review: If bit FIOM (Fence of I/O implies Memory) is set to one in menvcfg, FENCE instructions executed in modes [...]

Thanks~

closed 2021-11-04 by BiwefC

omerguzelelectronicguy, 2024-02-06:
I have a similar question. What if we didn't implement all the bits of `menvcfg` register? Should cpu take an illegal instruction exception? Or should it behave as WARL? 

I didn't understand 
> can simply ignore FIOM 
Ignoring means only storing written values but not using them for configuring devices or behaving as WARL?