riscv/riscv-isa-manual #765: mconfigptr should be called mconfigaddr

jrtc27, 2021-11-06:
A pointer is a language-level concept, the hardware construct is an integer address. People like to conflate the two, and for many architectures you can get away with it because they are effectively the same thing. In CHERI-RISC-V, they are not, pointers are no longer represented as pure integers but instead carry additional metadata including bounds and permissions. To more accurately capture the true meaning of the new CSR, and to not be a regrettable name should CHERI-RISC-V eventually be standardised, mconfigptr should instead be called mconfigaddr.

(I am aware we have [gst]p, but those are named as such because of the intended C ABI; it's mildly unfortunate for CHERI but so be it)

ted-tanner, 2023-12-10:
I agree completely that "addr" is better than "ptr" in this context. However, now that the spec has been hardened somewhat and Risc-V is shipping on real hardware, I doubt this will be changed.

kersten1, 2024-09-27:
Closing this one as name is still mconfigptr and as ted-tanner comments, I don't think it will change now.

closed 2024-09-27 by kersten1