riscv/riscv-isa-manual #769: Inconsistency between Privileged spec and Unprivileged spec related to counters

Silabs-ArjanB, 2021-11-12:
In the “Counters” chapter of Unprivileged ISA specification the following is mentioned:
 
“For a low-end implementation, the upper 32 bits of each counter can be implemented using software counters incremented by a trap handler triggered by overflow of the lower 32 bits.”

This seems to conflict with the fact that mcycleh and minstreth are mandatory in the Privileged Specification.

I have two questions related to this:

- It would be great indeed for low-end implementation if cycleh and instreth do not need to be implemented. This however only makes sense if the implementation is free to not implement mcycleh and minstreth either while still being compliant to the Privileged specification and this seems not to be allowed (as nowhere it is mentioned that an implementation is allowed to trap on accesses to mcycleh and minstreth). Is implementation of  mcycleh and minstreth optional (while still claiming compliancy to the Privileged ISA specification)? Even better, would it be allowed to trap on all of mcycle, mcycleh, minstret and minstreth CSRs while still claiming compliancy to the Privileged specification (as low-end implementations can easily do without any of these CSRs)?
- If we implement the mentioned ‘trap upon overflow’ for each of the counters, then it would be nice if the cause of such trap would get standardized in mcause. Are we supposed to just use some of the ‘designated for custom use’ values in mcause or are there plans to standardize these overflow bits. (I am aware of the proposed Sscofpmf extension, but that seems more related to the Zihpm extension as opposed to the Zicntr extension).

aswaterman, 2021-11-12:
The text you quoted is non-normative so shouldn't be taken as Gospel.

You're certainly right that this text is in conflict with the M-mode architecture, but note that a correct implementation of the unprivileged architecture technically doesn't require M-mode at all.

If you did take this approach, it would be a nonstandard feature, and so it would call for a nonstandard mcause code.

allenjbaum, 2021-11-13:
The conflict I see here is there is no architectural defined way to trap on
those counter overflows, and that trap isn't  invisible to Mmode - is that
the conflict you see?
I interpreted the "software counter" to mean a CSR that didn't have counter
hardware, as opposed to saving it in SRAM somewhere or something, ao
mcycleh, etc., would still exist.
I also realize that is silly, since they're objecting to more than just the
counter HW, but the flops s well.

aswaterman, 2021-11-13:
@allenjbaum I was thinking along those lines: the visibility of the trap to M-mode is the most problematic aspect, IMO.

Silabs-ArjanB, 2021-11-15:
Thank you for your answers. Agreed that 'that a correct implementation of the unprivileged architecture technically doesn't require M-mode at all'. This is why I added the phrase 'while still claiming compliancy to the Privileged ISA specification' in my question. The reason behind my question is indeed that it would be nice if there would be a way to eliminate the cost of the associated 128 flip-flops (mcycle(h) and minstret(h)) and 2 64-bit counters in small embedded cores, while still being compliant to both the Privileged and Unprivileged specifications. It is clear that this is not possible.

kersten1, 2024-09-27:
Closing as I don't think any more work needs to be done here. Please reopen if I am incorrect.

Silabs-ArjanB, 2024-09-30:
Hi @kersten1 , above discussion only raised more concerns than I posted myself in the original question. The (non-normative) quote that I asked about it still in Version 20240411 of the Unprivileged Architecture so I consider this issue to be unsolved.

gfavor, 2024-09-30:
@aswaterman It seems like the non-normative comment in the Unpriv spec (quoted up top) is not relevant in compliant implementations of the Machine-level Priv ISA?

If so, then should the non-normative comment be stricken or mpodified in some way?  If not, then is the idea that implementing the non-normative comment requires hardware support for an M-mode-invisible trap to a software emulation of the high halves of counter CSRs?  Or to what extent is treating this as a custom extension with use of a custom M-mode cause code still compliant with Machine-level Priv ISA?

aswaterman, 2025-12-09:
> For a low-end implementation, the upper 32 bits of each counter can be implemented using software counters incremented by a trap handler triggered by overflow of the lower 32 bits.

This non-normative note does more harm than good.  Clever implementers of low-end cores will figure this out themselves, and for everyone else, it is a disservice.  I'll make a PR to remove it.

closed 2025-12-09 by aswaterman