riscv/riscv-isa-manual #790: questions about SVNAPOT extension

leBsky, 2021-12-02:
The SVNAPOT chapter(chapter 5) of draft version 1.12 of RISC-V privileged doc states that when A PTE has N=1, the pte represents a translation that is part of a range of contiguous virtual-to-physical translation with the same value of PTE bits 5-0.

  My questions is :
   1.  How about other bits of PTE? I think these consecutive pages can be considered as another kind of superpage, so when in SVPBMT extension, the PTE[62:61] of these pages should also be identical.
   2.  Since these pages my be considered as a special superpage, so the PMP attributes(LRWX) should always be same for each page in this "superpage".  Is it correct?

JamesKenneyImperas, 2021-12-02:
The commentary says:

JamesKenneyImperas, 2021-12-02:
The commentary says:

_In typical usage scenarios, NAPOT PTEs in the same region will have the same attributes, same PPNs, and same values for bits 5–0. RSW remains reserved for supervisor software control. It is the responsibility of the OS and/or hypervisor to configure the page tables in such a way that there are no inconsistencies between NAPOT PTEs and other NAPOT or non-NAPOT PTEs that overlap the same address range._

So, yes, page table entries should be the same (but note the further information about A and D bits).

aswaterman, 2021-12-02:
Note the semantics are well defined if these constraints are violated (any of the various PTEs could be used to satisfy translations of any access to the NAPOT region), though essentially useless.

leBsky, 2021-12-03:
> Note the semantics are well defined if these constraints are violated (any of the various PTEs could be used to satisfy translations of any access to the NAPOT region), though essentially useless.

Thanks, It makes sense, and should these contiguous physical pages lie in a single PMP region?  My understanding is that since NAPOT PTE only regulates a virtual-to-physical translation, so it is allowed that the OS configures contiguous physical pages from different PMP regions into a NAPOT PTE.

JamesKenneyImperas, 2021-12-03:
PMP regions are independent of virtual memory pages, so the OS can legally configure PMP regions in the way you describe.

leBsky, 2021-12-07:
> PMP regions are independent of virtual memory pages, so the OS can legally configure PMP regions in the way you describe.

That's what I think, thanks.

closed 2021-12-07 by leBsky