riscv/riscv-isa-manual #795: possible bugs in figures A.20, A.21

aswaterman, 2021-12-15:
I'm copying this from @jrahmeh's issue here: https://github.com/riscv/riscv-v-spec/issues/765

In section A.7 of the unprivileged ISA (Document Version 20191214-draft), both examples (Figures A.20 and A21) seem to be incorrect.

In A.20, for outcome a0=1 to occur, a2 would have to be 1, but a2 is initialized to zero and remains zero after the amoswap at (e). One fix is to replace (h) with: sw t1, 0(s0)

In A.21, the outcome "a0=0, a1=1, a2=0" is not forbidden by operational model, it is obtainable with the global memory order: a b c d e f.
One possible fix is to change the outcome to: "a0=1, a1=1, a2=0" and change (f) to: sw t1, 0(s0)

skvadrik, 2022-09-23:
I am also confused by these examples. A few more notes:
- In A21, `t1` for hart 1 is unused (but it would be used with the fix fo (f) suggested by @jrahmeh)
- It would help to mention start conditions: I assumed all memory is initially zeroes, but the examples are confusing, so I was wondering if there were some special initial conditions not mentioned here.

hboehm, 2023-03-14:
It also seems likely to me that the A.20 and A.21 fixes suggested above are what was intended.

But A.22 is clearly also wrong. The outcome should pretty clearly state a2=1, not a1=1, which would be a contradiction. Since it works very hard to generate a syntactic address dependency on a2, I also expect the data dependency on a2 in (g) is not intended, and (g) should also mention t1 instead, as with the other examples.

I'll generate a pull request with what we believe the fixes should be.

aswaterman, 2023-03-14:
Thanks @hboehm!

PR #991 (2023-03-14): Fix Known Issues examples
fixes #795

Fixes are as suggested there by several contributors.
commit 8b9047d8 (2023-03-14): Fix Known Issues examples (#991)

fixes #795

Fixes are as suggested there by several contributors.
closed 2023-03-14 by aswaterman