riscv/riscv-isa-manual #798: A Typo in 'H' Extension Maybe?

retrhelo, 2021-12-18:
I'm recently reading _The RISC-V Instruction Set ManualVolume II: Privileged Architecture_, version 20211203. And I find it hard to understand at _Section 8.6.3_, which describes the possible values in a trap instruction CSR on a synchronous exception as below:

> On an interrupt, the value written to the trap instruction register is always zero. On a synchronous
> exception, if a nonzero value is written, one of the following shall be true about the value:
>
> -  Bit 0 is 1, and replacing bit 1 with 1 makes the value into a valid encoding of a standard instruction.
> ...
> - Bit 0 is 1, and replacing bit 1 with 1 makes the value into an instruction encoding that is explicitly
> designated for a custom instruction (not an unused reserved encoding).
> ...

It seems that the situation "Bit 0 is 1, and replacing bit 1 with 1" indicates two different types of instructions at the same time.

[image]

scottj97, 2021-12-18:
In the case that bit 0 is 1:

_If_ "replacing bit 1 with 1 makes the value into a valid encoding of a standard instruction" then the first bullet point applies.

_If_ "replacing bit 1 with 1 makes the value into an instruction encoding that is explicitly designated for a custom instruction" then the second bullet applies.

retrhelo, 2021-12-19:
I see, then the description makes sense for me. Thank you a lot for the explanation.

closed 2021-12-19 by retrhelo