riscv/riscv-isa-manual #799: a question about the "fence" instruction

jiamenghan, 2021-12-28:
Hi, all.
  I have a questio about the execution of fence instruction.

Given a instruction stream `sw data, (memory); fence iorw, iorw; sw flag (memory)`
Does the “fence” instruction itself ensure the pre-ordered `sw data, (memory)` have beed stored to the coherent point (such as dcache) ? or the completion of subsequent `sw flag,(memory)` ensure that the the data have beed stored to the coherent point? or only storing the flag to the coherent point ensuring that the data have beed stored to the coherent point?

It is meanful because it decides whether the CPU can still postpone the execution of previous st instructions when `fence` completes until the subsequent ld/st instructions being fenced appear.

aswaterman, 2021-12-28:
Ordering only, not completion.

jiamenghan, 2021-12-28:
Thx for the answer

closed 2021-12-28 by jiamenghan