riscv/riscv-isa-manual #800: Writing unsupported vsatp.MODE value

pdonahue-ventana, 2021-12-28:
> When V=1, vsatp substitutes for the usual satp, so instructions that normally read or modify
satp actually access vsatp instead.

When V=1 and we write to vsatp (using the encoding for satp) what should happen if the MODE value is unsupported?  If we were in S/HS and did a `csrw satp` then this statement is relevant:
> if satp is written with an unsupported MODE, the entire write has no effect; no fields in satp are modified.

If we were in S/HS and did a `csrw vsatp` then this statement is relevant:
> When V=0, a write to vsatp with an unsupported MODE value is not ignored as it is for satp.
Instead, the fields of vsatp are WARL in the normal way.

There is no guidance about writes to vsatp when V=1.  Since the field is WARL, presumably normal WARL rules apply.

However, I think that the "the entire write has no effect" rule should apply in this case.  Otherwise, this could be a virtualization hole (where carefully crafted operating system code could possibly detect that it was running in VS mode rather than S/HS mode).  And the explicit mention of V=0 WARL behavior leads me to believe that it was the intention all along to have V=1 act like satp.  If so, the vsatp description should be more explicit to avoid confusion.

@jhauser-us

scottj97, 2021-12-28:
The expected behavior you describe is the way Spike works.

aswaterman, 2021-12-28:
A guest running in VS-mode should see the same ISA interface as an OS running unvirtualized in S-mode, and so the rules for satp apply to vsatp when V=1.

pdonahue-ventana, 2021-12-28:
> A guest running in VS-mode should see the same ISA interface as an OS running unvirtualized in S-mode, and so the rules for satp apply to vsatp when V=1.

I agree that this is the way it ought to be.  I just don't think that the spec really says that, at least not somewhere that I can easily find (or somewhere that can be found by someone who reads the spec and implements the ISA but doesn't fully understand how virtualization works).

aswaterman, 2021-12-29:
Indeed, I'll add a clarification.

commit bc883ce9 (2021-12-29): Clarify V=1 [v]satp write behavior

Resolves #800
closed 2021-12-29 by aswaterman