riscv/riscv-isa-manual #814: Svinval: Which is correct HINVAL.[GV]VMA encoding?: ISA Manual or riscv-opcodes

a4lg, 2022-01-20:
I was synchronizing ISA Manual and [riscv-opcodes](https://github.com/riscv/riscv-opcodes) and found that those two have different encodings on `HINVAL.VVMA` and `HINVAL.GVMA` instructions.

Current ISA Manual (with manual quotation with `[]`):

    001[0]011 rs2 rs1 000 00000 1110011 HINVAL.VVMA
    011[0]011 rs2 rs1 000 00000 1110011 HINVAL.GVMA

Table generated by riscv-opcodes (with manual quotation with `[]`):

    001[1]011 rs2 rs1 000 00000 1110011 HINVAL.VVMA
    011[1]011 rs2 rs1 000 00000 1110011 HINVAL.GVMA

You can notice that difference between `[0]` and `[1]`.

Because of this, current GNU Binutils (compliant to the ISA Manual) and Spike (RISC-V ISA Simulator; compliant to riscv-opcodes) are incompatible.

Short question is: which encoding is correct? In either case, we need to fix relevant software.

If ISA Manual is correct:

* Fix riscv-opcodes
* Fix Spike (riscv-isa-sim)
* Fix other software which uses `encoding.h` generated by riscv-opcodes (if relevant)

If riscv-opcodes is correct:

* Fix ISA Manual
* Fix GNU Binutils

aswaterman, 2022-01-20:
The ISA manual is necessarily correct here--not just because it's the spec, but also because of the details of the encoding.  Bits 29:28 encode the minimum privilege for a SYSTEM instruction.  riscv-opcodes has those bits as 11, indicating an M-mode-only instruction, when in fact it's a hypervisor instruction.

I'll fix riscv-opcodes and Spike.  I think we can close this issue, since there's no problem with the spec.

aswaterman, 2022-01-20:
(BTW, thank you for finding this bug in riscv-opcodes!)

closed 2022-01-20 by aswaterman