riscv/riscv-isa-manual #817: hstatus.SPV should set to 0 on reset

jameshippisley, 2022-02-03:
I think that that hstatus.SPV bit should be reset to 0, and I don't see this stated in the spec. If hstatus.SPV is undefined on reset and does not get written by software, then it will be undetermined whether SRET puts the code into virtual mode or not. H-extension aware code can initialize hstatus.SPV during boot, but I believe that code which does not know anything about the H-extension and does not initialize the hstatus CSR should run correctly too.

jameshippisley, 2022-02-03:
I think the same issue applies for mstatus.MPV.

aswaterman, 2022-02-03:
My philosophy has generally been that M-mode software needs to know enough about the system it's running on to initialize this sort of state.  So, the only things that are required to be reset are the ones necessary for M-mode's functional correctness.

(In case you were wondering, the reset mandate about hgatp.MODE and vsatp.MODE is not for software's convenience; it's a functional issue, to prevent speculative execution of HLV/HSV in M-mode from causing page-table walks that touch unspecified addresses.)

jameshippisley, 2022-02-11:
Should I close this issue, or is that up to the spec owner?

aswaterman, 2022-02-11:
Always feel free to close your own issues if you think they’ve been suitably resolved. (I’ll close this one.)

closed 2022-02-11 by aswaterman