riscv/riscv-isa-manual #819: Should sstatus.MXR affect VS-stage translation from V-mode?

jameshippisley, 2022-02-11:
The spec states that:

"HS-level sstatus.MXR makes execute-only pages readable for both stages of address translation (VS-stage and G-stage), whereas vsstatus.MXR affects only the first translation stage (VS-stage)".

This makes sense for the hypervisor load/store instructions, and for MPRV=1 memory accesses. However, when a guest OS sets a page to execute-only, I believe it would never want that page to become readable when accessed from within the guest, unless it reprograms the page or its own MXR bit.

This is not a problem, provided the hypervisor always clears sstatus.MXR before entering the guest. However, this seems like an unnecessary programming requirement which increases the possibility of accidental or malicious bugs. So I wanted to raise the possibility that the above rule be enhanced to state that only VS-stage accesses initiated from non-V-mode are affected.

aswaterman, 2022-02-11:
The usage model is that sstatus.MXR is only ever briefly enabled then quickly disabled, so that “the hypervisor always clears sstatus.MXR before entering the guest” happens by construction. (In other words, there isn’t an additional programming requirement.) See for example here: https://github.com/riscv-software-src/riscv-pk/blob/2f3e6f530384b4ab9f8b39bceb8e86f2ba51a835/machine/unprivileged_memory.h#L76

aswaterman, 2025-12-09:
Closed as answered.

closed 2025-12-09 by aswaterman