riscv/riscv-isa-manual #820: state of menvcfg upon reset

atishp04, 2022-02-14:
The menvcfg CSR section in priv spec doesn't specify anything about the value at the reset. For some of the extension, the priv spec points to individual specifications(e.g. SSTC, CMO, SVPBMT) for the corresponding bits. However, those specification also do not say what should be the value of menvcfg after reset.

The corresponding bits in menvcfg can be set in M-mode but it needs to know whether the specific extension is available in hardware or not. For extensions with CSR support (e.g SSTC), it is bit easier to detect and set the menvcfg bit by read/write & trapping on that CSR.
The other extensions have to parse the DT to identify the presence of an extension or try setting menvcfg bit to check if it sticks.
However, all these would be unnecessary if hardware can set the correct bit in menvcfg upon reset.

Is that a viable approach ? If not, can we clarify the priv spec to specify that, menvcfg should have zero upon reset. It is the responsibility of the M-mode implementation to set it correctly.

aswaterman, 2022-02-15:
@gfavor was this an omission in the definition of each of the bits, or was leaving them unspecified the intent?

gfavor, 2022-02-15:
I've looked back over all the past envcfg-related emails and proposal drafts, and reset was never explicitly mentioned anywhere along the way.  So, in short, either no one thought about this or consciously the intent was that these would be initialized appropriately by M-mode code (like the many, many other registers that are not initialized by hart reset).  I believe the latter was the expectation, i.e. there wasn't a clear need to initialize menvcfg before being able to start executing instructions. 
 Consequently the issue being raised is not a matter of functional necessity

So early M-mode code needs to use the new "unified RISC-V low-level discovery method" being developed by tech-config to generally discover what extensions are implemented, what extension options are implemented, and what extension parameters are implemented.  Then that code needs to deal with determining whether to set or clear each bit, i.e. yet more information to be discovered.  Even though a given extension may be implemented, the associated enable bits may not want to be set (e.g. to provide backward compatibility).  Such as with the STCE and PBMTE bits.  And for other bits, like the CB*E bits, different systems may want different settings for these bits.

Hence it isn't clear what would be a universally appropriate definition of hart reset values that M-mode code could simply leave alone and not worry about whether they are suitable values for that system.  And suggesting that all bits corresponding to implemented extensions should be reset-initialized to ones as a discovery method is going in the opposite direction of where RV is trying to head with development of the new tech-config method.  (The *envcfg CSRs were specified not intended or architected to be a mechanism for doing extension discovery.)  Plus once one has discovered that info, one still has to also discover what is the **appropriate** setting for each bit.

At which point specifying a reset value of all zeroes doesn't seem to be of any value.  And one needs a discovery method that isn't solely based on whether a bit is writeable or not (because that won't tell you what is the appropriate value to initialize the bit to).

Lastly, given all this, there is no need for the Priv spec to single out this one CSR as needing to be initialized by M-mode code (versus all the other CSRs that need initializing).  All CSRs that don't have architecturally-defined values after a hart reset, all need to be initialized by software.

aswaterman, 2022-02-15:
Yeah, I think I agree.

In any case, it's still possible to probe whether a specific menvcfg bit is writable, even though menvcfg is not reset.  (Not that I'm saying it's a great discovery mechanism--I only mean to remark that not resetting the bits doesn't seem to preclude this strategy.)

atishp04, 2022-02-15:
Thanks for the explaining the rational behind the decision @gfavor 
Currently, the menvcfg CSR can be updated based on CSR detect or writing & reading the menvcfg for now.
Once the tech config parser is available, it can be used to set the menvcfg.

gfavor, 2022-02-16:
Just keep in mind that simply because certain bits in menvcfg are writeable, doesn't mean that they should be initialized to 1's.  Without some form of discovery of the appropriate values to initialize them to, simply setting them to 1's is not necessarily the appropriate thing to do.  Which leaves the M-mode code to be compiled with an understanding of the appropriate system-specific values to initialize menvcfg bits to (at which point discovering whether the CSR exists is most likely moot).

aswaterman, 2025-12-09:
Closing as answered.