riscv/riscv-isa-manual #824: Svnapot spec confusion

scottj97, 2022-02-22:
My team and I are confused by the [two bullet points](https://github.com/riscv/riscv-isa-manual/blob/35a8890d1cd0e53c408ef9fe2ce1014ce3b8d3e1/src/supervisor.tex#L2304-L2315) in the Svnapot chapter.

> If the encoding in _pte_ is valid according to Table 5.1, then instead of returning the original value of _pte_, implicit reads of a NAPOT PTE return a copy of _pte_ in which _pte.ppn_[_pte.napot_bits_ − 1 : 0] is replaced by _vpn_[_i_][_pte.napot_bits_ − 1 : 0]. If the encoding in _pte_ is reserved according to Table 5.1, then a page-fault exception must be raised.

The indexing of _pte.ppn_ here is apparently a bit index. (It doesn't make sense otherwise.) But that is different than the indexing of _pte.ppn_ elsewhere, including in the heading of Table 5.1 just above this paragraph, which uses _pte.ppn_[0] to refer to bits [18:10] of the PTE.

Also, what is the _i_ in _vpn_[_i_]? If that refers to the level, as _i_ does elsewhere, then the bit index doesn't make sense for anything except _i_=0.

I suppose you could write: _pte.ppn_[_i_][_pte.napot_bits_ - 9 * _i_ - 1  : 0] is replaced by _vpn_[_i_][_pte.napot_bits_ - 9 * _i_ - 1: 0] for all _i_ <= (_pte.napot_bits_  - 1) / 9. Or something crazy like that.

> Implicit reads of NAPOT page table entries may create address-translation cache entries mapping _a_ + _va.vpn_[_j_] × PTESIZE to a copy of _pte_ in which _pte.ppn_[_pte.napot_bits_ − 1 : 0] is replaced by _vpn_[0][_pte.napot_bits_ − 1 : 0], for any or all _j_ such that _j_[8 : _napot_bits_] = _i_[8 : _napot_bits_], all for the address space identified in _satp_ as loaded by step 0.

I believe what this second paragraph is trying to say is that implementations are permitted to cache a 64KB page as a series of 4KB pages, or any other in-between size. Yes? I base this on the commentary that follows which indicates that [software should do `sfence.vma`](https://github.com/riscv/riscv-isa-manual/blob/35a8890d1cd0e53c408ef9fe2ce1014ce3b8d3e1/src/supervisor.tex#L2341-L2346) on every 4KB chunk when a NAPOT PTE changes.

But I don't see how this paragraph leads to that result.

I assume that the _j_ here is analogous to the _i_ in Step 2 of 4.3.2, where _i_ refers to the tablewalk level and has a range of 0-2 in Sv39. How then is _j_[8 : _napot_bits_] relevant, since _napot_bits_=4? Those bits will always be 0.

It seems that _a_ + _va.vpn_[_j_] × PTESIZE will always point to a PTE. So maybe this paragraph has to do with caching intermediate (non-leaf) PTEs?

What is _i_ in this paragraph? There is only one reference to it. It must not refer to the tablewalk level, as _i_ does elsewhere. But I don't know what else it might be.

What is "step 0"? Section 4.3.2 has no step 0.

gfavor, 2022-02-23:
Unless Andrew or someone solidly answers this question, this should be directed to @daniellustig - who I believe wrote the text in question.

Insofar as the belief of what the second paragraph is saying - that is correct.  An implementation may cache individual 4KB pages, or a 64KB napot page, or anything in between.  And software can't assume anything in this regard and, since the semantics of sfence.vma were (intentionally) not changed, it must do an sfence.vma for each 4KB page within a 64KB napot page.

aswaterman, 2022-02-24:
It's true that since only 64 KiB NAPOT pages are currently defined, _i_ must be 0.  But Dan's formulation was generalized to handle NAPOT superpages (_i_ > 0), too.  I think generalizing the description was a defensible choice: should NAPOT superpages ever be defined, only the table need be altered.

_i_ is defined implicitly in Table 5.1 above, which is how we know _i_ must be 0 for the time being.

You are right that we were missing an `[i]` in `pte.ppn[i][pte.napot_bits − 1 : 0]`, and that `j` was misdefined (it should just be an integer 0..511, not an index into `vpn`).  And step 0 should be step 1.  Thanks for pointing these out; I've fixed them in 54eb7d664d6f0814a86c57971c8e5abcc3050915

I believe the remaining points are stylistic.

I'll leave this open so y'all and @daniellustig can check my work.

scottj97, 2022-02-24:
> It's true that since only 64 KiB NAPOT pages are currently defined, _i_ must be 0. But Dan's formulation was generalized to handle NAPOT superpages (_i_ > 0), too. I think generalizing the description was a defensible choice: should NAPOT superpages ever be defined, only the table need be altered.

I think my confusion is that I was assuming _napot_bits_ would continue growing, e.g. _napot_bits_=12 for page size of 16 MiB. If you define the 16 MiB page size with _i_=1 _napot_bits_=3 then it works.

I will have to think about the new paragraph 2. Am I right about what this is intended to say? (Storing 64KB as smaller pages is allowed.)

aswaterman, 2022-02-24:
> If you define the 16 MiB page size with _i_=1 _napot_bits_=3 then it works.

Right, that's how it would be defined.

> Am I right about what this is intended to say? (Storing 64KB as smaller pages is allowed.)

Yep, for example, a legal implementation of this extension is to ignore the N bit (modulo exception behavior, anyway).

daniellustig, 2022-02-24:
Thanks for the fixes @scottj97 @aswaterman.  Andrew's commit and explanations LGTM.

FWIW the table in the commentary part of the Svnapot spec also gives examples of other possible future encodings, including for superpages with i=1 (currently stopping just one row short of your 16MB example, but the pattern is there.)

closed 2022-02-24 by scottj97