riscv/riscv-isa-manual #825: Proposal for a common convention for synonyms for register ABI names in hand-coded assembler

scotws, 2022-02-23:
The sheer number of RISC-V registers can make it hard to keep track of what has been assigned where when coding assembler by hand. Some assemblers allow the creation of synonyms -- "renaming" -- for the ABI names of the registers via `.equ`, `.eqv` or a similar statement, which can help. Such synonyms can also reduce the chance that a single typo -- say, from `t0` to `t1` -- remains unnoticed by coder and assembler.

However, there is currently no standard or recommended best practice for this that I am aware of. Providing such a recommendation or standard could cut down on the number of variants in the wild, aiding readability and supporting debugging of hand-written code.

This issue proposes adding a simple, short recommendation for a "common convention" or "best practices" for creating synonyms for ABI register names when hand-coding assembler, for inclusion in _The RISC-V Instruction Set Manual Volume I: Unprivileged ISA,_ presumably in Chapter 25. The assumption is that this is going to happen anyway, so an attempt should be made to nudge coders towards a common format.

### Criteria

To keep it simple, there are three criteria for this scheme:

1. Synonyms should only be created if many registers are in use at the same time _and_ only for registers that are used for a longer time. In no way should "renaming" become the default practice.
2. The original ABI register name should be conserved to aid debugging. The idea is to _add_ information, accepting the inherent redundancy for the sake of human readers and error detection.
2. In addition, an indicator of the type of data stored in the register or its intended use may be included for the same reason.

### Syntax

**The original register ABI name is kept, but followed by an underscore; then an optional single letter for the type or use, followed by a second underscore; finally the user-chosen name.**

Examples:

```
    .eqv t0_n_cats t0      # current number of cats
    .eqv s1_p_catname s1   # pointer to the beginning of a string
    .eqv s2_f_havetuna s2  # flag to bool if goal achieved 
```

As an initial suggestion for the type indicators:

- **c** counter
- **f** flag
- **l** loop counter
- **n** number of
- **p** pointer

Important: The original ABI name of the register remains untouched and fully functional. Any new definition creates a synonym, not a replacement.

### Example: The Battle of the Four Armies

Consider as a toy example a battle between hobbits, orcs, elves, and dragons, whose numbers are stored in `s0, s1, s2`, and `s3` respectively. A code snippet to decide the winner could look like this:

```
        bgtz s3, dragons_win
        bgt s1, s2, orcs_win
        bgt s2, s1, elves_win
        j hobbits_win
```
Since these registers are used heavily in the rest of the code as well, we create synonyms following the scheme proposed: 

```
.eqv s0_n_hobbits s0  # number of hobbitses
.eqv s1_n_orcs s1     # number of orcs
.eqv s2_n_elves s2    # number of elves
.eqv s3_n_dragons s3  # number of dragons
```
Our code snippet then becomes:
```
        bgtz s3_n_dragons, dragons_win
        bgt s1_n_orcs, s2_n_elves, orcs_win
        bgt s2_n_elves, s1_n_orcs, elves_win
        j hobbits_win
```
The additional information makes the code more readable and the logic easier to follow while preserving all original information. Also, a typo such as `t2_n_elves` will now be caught by the assembler because this name is not defined.

### Drawbacks 

* Creating synonyms for register names or renaming registers is uncommon, since most processors simply do not have enough of them for these problems to occur. Any such change could cause confusion when readers first are confronted with code that contains renamed registers. Some people will object on principle. 
* There is currently no assembler-level support for this; some assemblers might even refuse creating register synonyms; possibly this might break some compiler/assembler level checks. 
* Register renaming might start to be considered a must or best practice in itself for hand-coding assembler, not a tool to be used sparingly in situations where a large number of registers in use becomes unwieldy. 
* There is currently no way to take back or delete synonyms once stated.

### Open questions

* A suggestion for the scope of using these modified names should be included, possibly on a per-subroutine or segment basis.

allenjbaum, 2022-02-23:
>    3. In addition, an indicator of the type of data stored in the

This would have help me a lot - but I would want to redefine the alias
whenever the current functionality is dead (often, in my code).
Without that, this is pretty useless to me.

pdonahue-ventana, 2022-02-23:
I've done this type of thing with #defines.  That allows redefinition (or undefinition) later in the test.

nick-knight, 2022-02-23:
Yeah, my team uses C preprocessor directives, like `#define`/`#undef`, extensively in our assembly codes. This is supported in both the GNU and LLVM assemblers. I would be interested to learn how we might benefit from the current proposal versus our preprocessor-based approach.

As an aside, I would suggest that assembly-language definitions and conventions be consolidated at https://github.com/riscv-non-isa/riscv-asm-manual , rather than to the ISA manual.

scotws, 2022-02-24:
Ah, sorry, thought this was the right place. Is there some way to move this issue over there @nick-knight or should I just close this and open a new issue there by hand?

aswaterman, 2022-02-24:
Closing the issue and reopening it there is the way to go.

scotws, 2022-02-25:
Moved to https://github.com/riscv-non-isa/riscv-asm-manual/issues/75 and closing here. Thanks!

closed 2022-02-25 by scotws