riscv/riscv-isa-manual #835: Add IPA pronounciation

piegamesde, 2022-04-19:
The spec opens with the following words (although it is not the only section with clarified pronunciation in the specs):

> RISC-V (pronounced “risk-five”)

This is only of limited use, especially for non-native speakers: "The spelling of an English word tells you nothing about its pronunciation, but everything about its history". A much better alternative would be to use the IPA alphabet to write down how exactly it is pronounced.

kasanovic, 2022-04-19:
Please submit a PR.

kbroch-rivosinc, 2024-07-16:
From https://www.internationalphoneticalphabet.org/english-to-ipa-translator/

```
( risk : /ˈɹɪsk/ ) ( five : /ˈfaɪv/ )
```

IMO the "risk-five" is simply trying to avoid the confusion of users saying "risk-vee"

I do think that if/when the spec is translated to say German, one could expect English "words" to have the IPA pronunciation like https://de.wikipedia.org/wiki/RISC-V

aswaterman, 2025-12-09:
Closing, but feel free to submit a PR in future.