riscv/riscv-isa-manual #841: Please specify the interaction of JALR offsets with respect to the RAS .

karlosz, 2022-04-29:
For the riscv backend of the compiler sbcl(http://www.sbcl.org/), I have made the choice of using JALR offsets to call tagged function pointers quickly, by using the offset to untag the pointer., like so `jalr x2, tagged-function-pointer, tag`. Obviously `x2` should just get pushed to the RAS as is,, and there's no penalty for using an offset with prediction in this case.

However, tagged return address are a way for languages with moving GCs to relocate on-stack precisely. This is a slightly contrived example, but illustrates the point. Let's assume the return address tag is 5, and say the call sequence is:
```
  jal x2, foo
  ....
foo:
  addi x5, x2, +5
``

Let's say at the end of this sequence, x2 = 20, x5 = 25. Then the top of the RAS will be 20.

Assume the body of foo doesn't touch x5, but x2 has been destroyed. Then we could have a return address sequence look as follows.

```jalr x0, x5, -5````

It's not specified how this instruction would interact with the RAS.  Would the effective address that gets computed be `x5 - 5`, thus counting as a "cache" hit? Or would x5 be used? Does such code incur a penality? A clarification in the specification about the interaction would be nice.

kasanovic, 2022-04-29:
This is a micro architectural detail, but RAS designs will always track/check the effective return address not the operands.

karlosz, 2022-04-29:
Thank you. That's what I expected from the hardware point of view, especially as software would not be able to take advantage of the offset scheme as much otherwise, but I think it would be worth a note somewhere to clarify what software can expect.

scottj97, 2022-04-29:
I don't see what needs clarification. If an instruction pops the RAS then the hardware will compare the predicted destination address (from the RAS) with the actual computed destination address. As long as they match, there will be no misprediction.

This is what a RAS does. Maybe I'm missing something but I don't see how it could work any differently.

kasanovic, 2022-04-29:
I agree with @scottj97 
Adding clarification would be equivalent to saying "assume the RAS microarchitecture is not broken".  
Closing.

closed 2022-04-29 by kasanovic