riscv/riscv-isa-manual #846: Ambiguity about *tval in case of virtual instruction exception

Phantom1003, 2022-05-16:
> On a virtual instruction trap, mtval or stval is written the same as for an illegal instruction trap.

The specification said it is the same as the illegal instruction trap, the word `same` is not very clear.
1. Since mtval is either set to zero or written with illegal instruction on the illegal instruction trap, so mtval is also either set to zero or written with illegal instruction on the virtual instruction trap.
2. Or the value in mtval must be the same as the illegal instruction when a virtual instruction exception occurs. In this view, we can consider this as converting a virtual instruction exception to an illegal instruction exception, except that the Trap Cause Codes remain unchanged.

I would like to confirm which understanding is correct, thanks.

scottj97, 2022-05-16:
Are you asking if it's okay for an implementation to write the illegal instruction's opcode bits into `mtval` on an illegal instruction trap, and write 0 into `mtval` on a virtual instruction trap?

Phantom1003, 2022-05-16:
Yes, that's right.

gfavor, 2022-05-16:
The quoted Priv spec sentence, read literally, means that mtval/stval is set on an virtual instruction trap in exactly the same way as for an illegal instruction trap.  Inclusion of the word "exactly" just emphasizes the point being stated by the quoted sentence.

But still, having said that, one could try and apply a non-literal reading and say that it is ok to implement different behavior for illegal versus virtual instruction traps.

My own suggestion would be to not go there, and to implement the same behavior in both cases.  But if someone feels the need to implement differently and hence needs to ask whether the non-literal reading of the quote is architecturally valid, then someone like John Hauser (author of the H extension spec) will need to weigh in on the architectural intent of the quoted text.

allenjbaum, 2022-05-16:
My pleas would be to not go there also.
An implementation that does that would like not pass architectural tests
(when they're eventually written) and would have to apply for a waiver to
certify that it's compatible

Phantom1003, 2022-05-17:
cc to @jhauser-us

jhauser-us, 2022-05-17:
> Are you asking if it's okay for an implementation to write the illegal instruction's opcode bits into `mtval` on an illegal instruction trap, and write 0 into `mtval` on a virtual instruction trap?

Is that the question?

As I read it, the RISC-V Privileged Architecture doesn't say that one of these two statements must be true:
- `mtval`/`stval` is always set to zero on every illegal instruction trap; or
- `mtval`/`stval` is always written with the faulting instruction on every illegal instruction trap.

Instead, I believe the spec says that an implementation is free to make an independent choice (zero or not zero) on each illegal instruction trap.  Hence, it should also be okay for an implementation to write the faulting instruction into `mtval`/`stval` on every illegal instruction trap, and set `mtval`/`stval` to zero on every virtual instruction trap, because that's a valid choice (zero or not zero) for each trap.

Be aware, though, that RISC-V platform standards may make additional rules that reduce an implementation's freedom.

If that's not really the question, then I don't understand the question, to be honest.

Phantom1003, 2022-05-17:
Yes, that's the point I want to ask. The original description gives the impression that it must write the same value as the illegal instruction trap. But according to your explanation, they should be independent and have the same policy of being free to choose the value to write (zero or illegal instruction).

And thank you all, my question is solved. If there is nothing else to discuss, I will close this issue.

allenjbaum, 2022-05-17:
To say it another way (because I interpreted that statement differently,
and clearly incorrectly):
 for any specific value in the xcause CSR, one of those statements must be
true, but it can vary from xcause value to xcause value.
I.e if xcause=2 (illegal instruction) it might always store the faulting
instruction, but for cause=22 (virtual instruction) is might always set it
to zero -
and not that it can sometimes store the faulting instruction and sometimes
zero for illegal instruction, xcause 2.

I assumed the above, but  interpreted the spec statement to mean that it
wasn't mode dependent as well:
i.e. if Umode stored the instruction on an illegal instruction exception,
then
every mode that took an illegal instruction exception would also store the
instruction.

closed 2022-05-18 by Phantom1003