riscv/riscv-isa-manual #848: Question about MRET

carlocaione, 2022-05-24:
In the unprivileged spec we read:
```
We use the term trap to refer to the transfer of control to a trap handler caused by either an
exception or an interrupt.
```
About `mret` in the privileged spec we read:
```
An MRET or SRET instruction is used to return from a trap in M-mode or S-mode respectively.
```
and
```
The MRET instruction is used to return from a trap taken into M-mode.
```

My question is: **is it legal to call `mret` when not in a trap handler?**

Let's say for example that I'm already running in `m-mode` (for example I'm running some RTOS in m-mode), is it legal from the spec prospective calling `mret` to jump to `mepc` always staying in m-mode?

scottj97, 2022-05-24:
> My question is: **is it legal to call `mret` when not in a trap handler?**

Yes. Being "in a trap handler" is a convenient term for software. The architecture doesn't know or care if it's "in a trap handler" at any given moment. Traps will write a bunch of CSRs and redirect program execution; `mret` will read a bunch of CSRs and redirect program execution. As long as you set the CSRs appropriately, `mret` will do what you tell it.

tovine, 2022-05-24:
> The architecture doesn't know or care if it's "in a trap handler" at any given moment.
But it will know from the `mcause` value?

Also, what's the expected behavior if repeatedly calling `mret` then - simply jumping to `mepc` and setting `mstatus.mie` to `mstatus.mpie` (and any other CSRs that are to be restored)?
And then doing the exact same thing over and over again (if one keeps on calling `mret`)?
Doesn't intuitively seem like what the instruction was designed for

kasanovic, 2022-05-24:
The spec describes the behavior of the hart when executing `mret`.  This spec does not refer to `mcause`, and is no effect on `mret`.  Repeatedly executing `mret` will just repeatedly follow the given spec.  The MPP/MPIE stack will be emptied and because top of stack MPP is set to user mode, the next time trying to execute MRET will cause an illegal instruction trap.

scottj97, 2022-05-24:
> But it will know from the `mcause` value.

There's nothing for it to "know". You can `csrw mcause` and put whatever you want there at any time (in M-mode), and as Krste said, `mret` doesn't care about `mcause`.

> Also, what's the expected behavior if repeatedly calling `mret` then - simply jumping to `mepc` and setting `mstatus.mie` to `mstatus.mpie` (and any other CSRs that are to be restored)? And then [...]

Like any instruction in the ISA, `mret` will do what the spec says each time it's executed. You're right, it wasn't designed to be called repeatedly -- so why would you do such a thing?

closed 2022-07-21 by carlocaione