riscv/riscv-isa-manual #854: Accessing Virtual memory from M-mode

konrad-schwarz, 2022-06-14:
More complex SEE interfaces (more complex than the initial set of interfaces defined in the original SBI) will want to access S-mode virtual memory, to make the S-mode's life easier, by not requiring buffer address arguments to be translated into physical addresses.

The H-Extension introduces specialized load/store instructions to access the guest's virtual memory (and its physical memory by temporarily shutting off translation in the vsatp CSR), which aids paravirtualization, which is a similar use case to the SEE.

Would it be possible to extend these instructions so that they use S-mode translation when the pre-exception V is off and transfer the instructions to the base privileged instruction set?

Otherwise, either S-mode or M-mode code will need to explicitly convert to physical addresses, which could be expensive and is prone to races (and associated attacks of S-mode onto M-mode code), also, buffers might not be contiguous in physical memory.

aswaterman, 2022-06-14:
This proposal would be an incompatible change to a ratified standard.

In the current standard, you can accomplish something similar using MPRV + regular loads and stores.

closed 2022-06-14 by konrad-schwarz

aswaterman, 2022-06-14:
👍