riscv/riscv-isa-manual #864: [question] misaligned instruction exception if C extension enabled

stnolting, 2022-07-02:
Chapter 18 "_C Standard Extension for Compressed Instructions, Version 2.0_" of the RISC-V Instruction Set Manual (unprivileged ISA) says:

> With the addition of the C extension, no instructions can raise instruction-address-misaligned exceptions.

My question is: why?

On a `rv32i` architecture this will cause an _misaligned instruction address_ exception, right?
```
li x1, 3
jr x1    -> exception because x1[1] != 0
```

What happens on a `rv32ic` when doing a _jump to register_ instruction where bit 0 of the source register is set? E.g.:
```
li x1, 3
jr x1    -> ???
```

Shall the LSB be ignored entirely by the CPU implementation? So in this case the CPU would just jump to address 2?

aswaterman, 2022-07-02:
The JALR instruction does indeed ignore the LSB of the effective address, so in your example, the result will be a jump to address 2. https://github.com/riscv/riscv-isa-manual/blob/ca0a010494847c841660e92131b0add2593b2cf6/src/rv32.tex#L793-L805

stnolting, 2022-07-03:
Thank you very much! That was exactly what I was looking for.

I guess this
> Although there is potentially a slight loss of error checking in this case, in practice jumps to an incorrect instruction address will usually quickly raise an exception.

also applies to _**all**_ other situations including this one (`rv32ic`, executed in machine-mode):
```
li   x1, 3
csrw mepc, x1
mret           -> no exception, just jumping to address 2 (ignoring destination's LSB)
```

Correct?

aswaterman, 2022-07-03:
mepc[0] is always zero, so jumping to an odd-numbered address upon MRET is also impossible.

Anyway, the text you originally quoted is correct: misaligned instruction address exceptions are impossible if the C extension is implemented.

stnolting, 2022-07-04:
Great! Thanks for clearing!

closed 2022-07-04 by stnolting