riscv/riscv-isa-manual #868: Confusion on interrupt-related descriptions of privilige spec

jerryhethatday, 2022-07-12:
**In the 3.1.9 section of riscv-privileged spec(mie and mip):**

> An interrupt i will be taken if bit i is set in both mip and mie, and if interrupts are globally enabled.By default, **M-mode interrupts** are 
>  globally enabled if the hart’s current privilege mode is less than M, or if the current privilege mode is M and the MIE bit in the mstatus register is set. If bit i in mideleg is set, however, interrupts are considered to be globally enabled if the hart’s current privilege mode equals the delegated privilege mode (S or U) and that mode’s interrupt enable bit (SIE or UIE in mstatus) is set, or if the current privilege mode is less than the delegated privilege mode.

I have confusion on the specific definition of so-called **M-mode interrupts** , does this mean interrupts like MEIP, MSIP, MTIP? Or, this  **M-mode interrupts** just mean all the interrupts that **is not** delegated to lower  privilege levels? 

**In 3.1.6.1: Privilege and Global Interrupt-Enable Stack in mstatus register**

> When a hart is executing in privilege mode x, interrupts are globally enabled when x IE=1 and
> globally disabled when x IE=0. **Interrupts for lower-privilege modes**, w<x, are always globally
> disabled regardless of the setting of the lower-privilege mode’s global w IE bit. Interrupts for
> higher-privilege modes, y>x, are always globally enabled regardless of the setting of the higherprivilege
> mode’s global y IE bit. Higher-privilege-level code can use separate per-interrupt enable
> bits to disable selected higher-privilege-mode interrupts before ceding control to a lower-privilege
> mode.
The description above also confuses me a lot, does **Interrupts for lower-privilege modes** just means SEIP, SSIP, STIP(in terms of S-MODE), Or, it just means **Interrupts  that **is** delegated to lower  privilege levels**?

Also, **In 3.1.8 Machine Trap Delegation Registers (medeleg and mideleg)**:

> Delegated interrupts result in the interrupt being masked at the delegator privilege level. For
> example, if the supervisor timer interrupt (STI) is delegated to S-mode by setting mideleg[5], STIs
> will not be taken when executing in M-mode. By contrast, if mideleg[5] is clear, STIs can be taken
> in any mode and regardless of current mode will transfer control to M-mode.

From description at 3.1.8, I can infer that the  **M-mode interrupts** in 3.1.9  just means all the interrupts that is not delegated to lower  privilege levels?  And  **Interrupts for lower-privilege modes**  at 3.1.6.1 means **Interrupts  that is delegated to lower  privilege levels**.  But I am not confident this is true.
Can anyone shed some light into this? For example, if an SSIP is not delegeated to S-mode, will this interrupt be taken in M-mode？
Thanks!

aswaterman, 2022-07-12:
> From description at 3.1.8, I can infer that the **M-mode interrupts** in 3.1.9 just means all the interrupts that is not delegated to lower privilege levels? And **Interrupts for lower-privilege modes** at 3.1.6.1 means **Interrupts that is delegated to lower privilege levels**.

Yes.

jerryhethatday, 2022-07-12:
> > From description at 3.1.8, I can infer that the **M-mode interrupts** in 3.1.9 just means all the interrupts that is not delegated to lower privilege levels? And **Interrupts for lower-privilege modes** at 3.1.6.1 means **Interrupts that is delegated to lower privilege levels**.
> 
> Yes.

Thanks for your replay. :)

closed 2022-07-12 by jerryhethatday