riscv/riscv-isa-manual #873: [question] "Reserved" mcause value: Interrupt = 1, Exc. Code = 0

stnolting, 2022-07-22:
Table "3.6: Machine cause register (mcause) values after trap" in section "3.1.15 Machine Cause Register (mcause)" lists the _Interrupt_ and _Exception Code_ bit-field combinations of `mcause` defined so far:

[image]

The combination Interrupt = 1 and Exception Code = 0 is marked as "reserved". According to section "3.5 Non-Maskable Interrupts" this combination is not "reserved" (= "yet unused, might be assigned in future") but used to identify an unknown interrupt source:

> The values written to mcause on an NMI are implementation-defined. The high Interrupt bit of mcause should be set to indicate that this was an interrupt. **An Exception Code of 0 is reserved to mean “unknown cause” and implementations that do not distinguish sources of NMIs via the mcause register should return 0 in the Exception Code.**

So maybe the wording in table 3.1.15 should be changed to "unknown source" to make clear this Interrupt/Exception Code combination is actually _defined_? Or is this specific combination actually undefined - and just defined if the platform supports NMI?

In general the wording 
> An Exception Code of 0 is reserved to mean “unknown cause”

might be misleading, because an _Instruction address misaligned_ exception also uses Exception Code = 0 but this case is a defined source.

Thanks for any help in advance! 😉

allenjbaum, 2022-07-22:
This paragraph that has
    An Exception Code of 0 is reserved to mean “unknown cause”'
is referring explicitly to the exception code written as the result of an
NMI, and is not referring to exceptions.
Exception code field of the xCAUSE CSR has 3 separate meanings, depending
on the int bit (MSB) and (when the int bit is set)
whether the trap vector was taken from xTVEC or the implementation defined
NMI vector
So, I'd argue that the table might want to make it clear that these are for
regular interrupts, and not NMIs,
but ALL encodings (not just zero) in the exception code field in that table
are incorrect for NMIs.

stnolting, 2022-07-22:
Understood! Thanks for clearing! :+1:

closed 2022-07-22 by stnolting