riscv/riscv-isa-manual #878: Confusion about meaning of zero in mideleg

dramforever, 2022-08-01:
I'm trying to understand what use *not* delegating a supervisor interrupt could have. As an example, what extra things could M-mode software do if `mideleg.SEI = 0`?

I understand that *not* delegating exceptions gives M-mode a chance to 'hijack' them before sending them to S-mode, but this does not appear to be possible with `mideleg`.

The spec says:

>  Restricted views of the mip and mie registers appear as the sip and sie registers for supervisor level. If an interrupt is delegated to S-mode by setting a bit in the mideleg register, it becomes visible in the sip register and is maskable using the sie register. Otherwise, the corresponding bits in sip and sie appear to be hardwired to zero.

So it wouldn't be possible to properly raise a supervisor external interrupt in software from M-mode software because the enable/pending bits just stop working in S-mode. To S-mode software it is as if SEI isn't implemented.

So it seems that the only thing not delegating an interrupt can do is to hide its existence from lower-privileged software? With no possibility for injecting them manually? This is very confusing. Why does interrupt delegation, or rather the lack thereof behave in this seemingly useless way?

(This was also posted on Reddit at <https://www.reddit.com/r/RISCV/comments/wcr7um/confusion_about_meaning_of_zero_in_mideleg/>)

aswaterman, 2022-08-01:
It is indeed the case that not delegating supervisor interrupts to S-mode is not often useful.  Emulating less-capable systems (i.e. those that lack that particular interrupt) is one use case, though.  Consider the case of paravirtualizing a system that doesn't actually have an external interrupt controller, but uses some alternate mechanism for receiving that sort of interrupt.

There is also a notional simplification in making all interrupts identical: they all go to M-mode unless delegated, no matter what their purpose is.  But I acknowledge the value of this property is in the eye of the beholder.

scottj97, 2022-08-01:
Isn't it still be possible to trigger SEI via M-mode `csrw mip`? It would go to the M-mode handler with `mcause` of 0x80000009.

Maybe you're saying that's not useful since `sip` and `sie` are hardwired to 0?

dramforever, 2022-08-02:
> Maybe you're saying that's not useful since `sip` and `sie` are hardwired to 0?

Indeed, this is the bit that was the most confusing for me.

If I set an exception to be non-delegated in `medeleg`, after trapping in M-mode, M-mode software has the option to emulate an exception trap to S-mode. If I set an interrupt to non-delegated in `mideleg`, it traps in M-mode but there's no way to emulate an interrupt trap to S-mode, since it couldn't be enabled in S-mode.

It seems that exceptions can 'go through' from M-mode to S-mode fine, but interrupts once trapped in M-mode cannot actually 'go through' to S-mode. Is this the case? For a start, it's not even possible to tell if S-mode currently has a particular non-delegated interrupt enabled, because `sie` would be read-only zero in that bit. It's also not possible to trap accesses to `sie` and `sip` without completely classically virtualizing S-mode in U-mode.

aswaterman, 2022-08-02:
You're right that the feature isn't sufficient for M-mode to intercept then emulate interrupts destined for S-mode without additional mechanisms (but then again, no one claimed it was).

dramforever, 2022-08-02:
Okay, so I guess it's just not really designed to work the same as delegating exceptions. That's fair enough to me.

closed 2022-08-02 by dramforever