riscv/riscv-isa-manual #881: Help to clarify expected behavior of mtval?

ma142857, 2022-08-11:
https://github.com/riscv/riscv-isa-manual/blob/190ed91590f6f58f3ac3236a8c7081aec1643ae0/src/machine.tex#L2246
In the mtval chapter, the Privileged spec states that:
> If mtval is written with a nonzero value when an instruction access-fault or page-fault exception occurs on a system with variable-length instructions, then mtval will contain the _**virtual address**_ of _**the portion of the instruction**_ that caused the fault, while mepc will point to the beginning of the instruction.

Does the **_virtual address_** here really means effective address? Since instruction access-fault exception could occurs in bare mode.

And could you help to confirm e.g., when an unaligned 4-byte instruction fetch is decomposed into 2 accesses:
1. In Bare mode, if the 1st access passes PMP while the 2nd access fails PMP. Does mtval expected to be different than mepc (mepc+2 == mtval)?
2. In MMU mode, if the 1st access passed MMU and PMP while the 2nd access passes MMU but fails PMP. Does mtval expected to be different than mepc (mepc+2 == mtval)?

Thanks in advance!
Andy

gfavor, 2022-08-11:
On Thu, Aug 11, 2022 at 7:40 AM Andy Ma ***@***.***> wrote:

> In the mtval chapter, the Privileged spec states that:
>
> If mtval is written with a nonzero value when an instruction access-fault
> or page-fault exception occurs on a system with variable-length [...]
Yes.

> And could you help to confirm e.g., when an unaligned 4-byte instruction
> fetch is decomposed into 2 accesses:
>
>    1. In Bare mode, if the 1st access passes PMP while the 2nd access
>    fails PMP. Does mtval expected to be different than mepc (mepc+2 == mtval)?
>
> Yes - per the above quoted excerpt from the Priv spec.

>
>    1. In MMU mode, if the 1st access passed MMU and PMP while the 2nd
>    access passes MMU but fails PMP. Does mtval expected to be different than
>    mepc (mepc+2 == mtval)?
>
> Yes.  PMP checks are agnostic to what happened upstream (as far as where
the PA came from and how it was produced/calculated).

Greg

Message ID: ***@***.***>
>

ma142857, 2022-08-12:
Thanks @gfavor for the clarification!
It will be good if the Privileged Spec could be updated to use _**effective address**_ instead of _**virtual address**_. 

Best,
Andy