riscv/riscv-isa-manual #882: mstatus.MPIE bit and MRET

algrobman, 2022-08-12:
what is it good for that MRET sets mstatus.MPIE bit? 

with this behavior, two MRET instructions in row set mstatus.MIE bit, enabling interrupts, which is feels like unexpected result.
(sequence may happen if an exception handler takes another exception).

aswaterman, 2022-08-12:
Two MRET instructions in a row is going to mess things up for other reasons (namely, it'll change the privilege mode to U, no matter what software intended), so the behavior in that case isn't especially interesting.  Additional software care is necessary to take exceptions within trap handlers; changing the definition of MPIE wouldn't suffice.  With that in mind, we chose to define MRET to set MPP and MPIE to values that will cause such software bugs to be caught quickly.

algrobman, 2022-08-12:
Other architectures with similar exception/interrupt handling mechanism (PowerPC MSR/SRR0/1 as example) leave "previous" mode/IE state unchanged by Return from Interrupt instructions. Thus nested exceptions/interrupts return CPU to the same state as it was before these exceptions. which is more reasonable behavior, I think, than having some hidden side effects, programmer need to take care of if wants to avoid them. (Of course to support nested exceptions some additional code to be written to save/restore these "previous" state requisites in/from stack, but this is more traditional/clear way of doing things in exception handlers, than redoing what HW did  ... )