riscv/riscv-isa-manual #883: Missing CSR address for `mtime`

ObiWanRohan, 2022-08-16:
Chapter 3 , Section 3.2.1 defines that a `mtime` register should be present, but the CSR address for this register isn't present anywhere in the spec.

The address was present in the spec for v1.7(In the CSR Listing table), but it seems to have been removed.
Is the address for this register user-defined now, or am I misreading something?

aswaterman, 2022-08-16:
It isn't a CSR.  It's a memory-mapped register (whose address is platform-dependent).

stnolting, 2022-08-16:
@ObiWanRohan 

Btw, there is a `time` CSR (or two CSRs `time` and `timeh` on rv32, respectively):
> The time CSR is a read-only shadow of the memory-mapped mtime register. Analogously, on RV32I the cycleh, instreth and hpmcounternh CSRs are read-only shadows of mcycleh, minstreth and mhpmcounternh, respectively.
[3.1.11 Machine Counter-Enable Register (mcounteren)]

ObiWanRohan, 2022-08-17:
Thanks @aswaterman @stnolting  for the clarifications.

mickflemm, 2023-04-30:
There is still a mention regarding mtime in section 3.1.10, what is the bit TM of `mcounteren` supposed to do since mtime is outside the hart ? ignore timer interrupts ?

gfavor, 2023-05-01:
>
> There is still a mention regarding mtime in section 3.1.10, what is the
> bit TM of mcounteren supposed to do since mtime is outside the hart ?
> ignore timer interrupts?
>
Doesn't the priv spec clearly specify what the TM bit does (which is
orthogonal to the fact that mtime comes from outside a hart and that the
CSR in question is the 'time' CSR - which is a typically delayed version of
mtime) when it says the following?

*When the TM bit in the mcounteren register is clear, attempts to read the
time register while executing in S-mode or U-mode will cause an illegal
instruction exception. When one of these bits **[TM] **is set, access to
the corresponding register [time] is permitted in the next implemented
privilege mode (S-mode if implemented, otherwise U-mode).*

Btw, you also appear to be looking at a not-latest version of the Priv spec
(not that that matters for this question).

Greg

mickflemm, 2023-05-01:
I get what it does, I don't understand what prevents S-mode from accessing `mtime` directly since it's memory mapped, this mostly seems as a way to trap-and-emulate accesses to `time` on M-mode, it's not really about accessibility as the first paragraph mentions ("The settings in this register only control accessibility") since `mtime`, the source of `time` (unless blocked by PMP for example) is still accessible.

allenjbaum, 2023-05-01:
Well, the OS level might not know the address (since it is intended to be
Mmode accessible only), but blocked by PMP or some priv level sensitive PMA
is the only other option.

gfavor, 2023-05-02:
> I get what it does, I don't understand what prevents S-mode from accessing
> mtime directly since it's memory mapped, this mostly seems as a way to
> trap-and-emulate accesses to time on M-mode, it's [...]
'mtime', as a memory-mapped register, is just like all other memory-mapped
registers that M-mode wants to prevent lower modes from having access to,
i.e. M-mode uses PMPs to restrict access to mtime and all other such MMRs.

tovine, 2025-02-28:
Sorry for reviving this thread after such a long time, but the spec document is still a bit ambiguous on this topic.

I searched through the archives and came across [this email thread](https://lists.riscv.org/g/tech-privileged/topic/requirements_on_implementing/96375353), where the discussions seemed to conclude that even if you claim to implement the `Zicntr` extension, it is still ok to trap any accesses to the `time` CSR (even in M-mode) and read the memory mapped (platform-defined) real-time counter into the destination of that CSR access.

It would be very nice if this could be confirmed and put into writing as a note in the spec documents, as I'm not sure whether or not I can include `Zicntr` as a supported extension if my implementation has the `mcycle[h]` and `minstret[h]` but raises illegal instruction when reading `time`.