riscv/riscv-isa-manual #884: Clarification of PMP grain effect on TOR region lower bound

JamesKenneyImperas, 2022-08-23:
The Privileged Specification says this about address masking for TOR PMP regions:

> When `G ≥ 1` and `pmpcfgi.A[1]` is clear, i.e. the mode is OFF or TOR, then bits `pmpaddri[G-1:0]` read as all zeros.

In the case of TOR mode, is it also the case that `pmpaddr(i-1)[G-1:0]` read as all zeros? If so, could this be explicitly stated?

Thanks.

aswaterman, 2022-08-23:
Nope, it means what it says.  (I can't say at the moment I think the definition is terribly useful, but I don't think it's ambiguous.)

JamesKenneyImperas, 2022-08-23:
I realise that I didn't specify what I meant clearly enough - I don't mean an explicit read of `pmpaddr(i-1)` by a CSR operation, but how the value of `pmpaddr(i-1)` is treated for address match purposes. There is a second statement that says:

> Bits `pmpaddri[G-1:0]` do not affect the TOR address-matching logic.

Should this also include `pmpaddr(i-1)[G-1:0]` for a TOR entry? This section talks about the effect of G on entry `i`, but says nothing about entry `i-1`.