riscv/riscv-isa-manual #885: Unclear abbreviation `GC extension`

goekce, 2022-08-25:
I am reading the manual for the first time and could not understand what `GC` stands for:

https://github.com/riscv/riscv-isa-manual/blob/10eea63205f371ed649355f4cf7a80716335958f/src/intro.tex#L352-L352

Is this a bug or feature?

ethindp, 2022-08-25:
`GC` means "General with Compressed", I think. The RISCVxG extension contains the I, A, M, F, and D instruction sets. The `C` part of that acronym means that the `C` instruction set extension is added for compressed instructions.

goekce, 2022-08-25:
Makes sense, thanks!

If you are right, then `G` should be defined in the intro to avoid confusion. What do the authors think?

aswaterman, 2022-08-25:
Agreed, the forward reference to G probably resulted from years of editing.  Fixed.

commit 1237e31f (2022-08-25): Avoid forward reference to "G" in intro

Resolves #885
commit abb52ae4 (2022-11-04): Avoid forward reference to "G" in intro

Resolves #885
commit 423c3d36 (2022-11-04): Avoid forward reference to "G" in intro

Resolves #885
closed 2022-08-25 by aswaterman