riscv/riscv-isa-manual #889: Ambiguous expression `following the JAL`

goekce, 2022-08-30:
https://github.com/riscv/riscv-isa-manual/blob/7859a79531ac1dd43c696c46f81e2202ad453678/src/rv32.tex#L599-L600

I understand that `JAL +4` will jump to the next instruction after itself (in the subroutine) and store the address of the instruction following the jump in `rd` according to:

https://github.com/riscv/riscv-isa-manual/blob/7859a79531ac1dd43c696c46f81e2202ad453678/src/rv32.tex#L716-L717

So if `JAL +4` instruction is at address $x$, then this instruction will jump to $x+4$, and store $x+8$ in `rd`. Am I right?

If I am right, then the expression `following the JAL` could be understood both ways:
1. The instruction that follows the JAL instruction $x+4$
2. The instruction following the JAL (jump) operation. I jump first and I am at $x+4$. The following operation would be $x+8$.

In this case I would stick to `following the jump` as in other sections.

What do you think?

goekce, 2022-08-30:
On a second thought I think I am wrong. It does not make sense for `JAL` to store an address after the jump operation, because the address $x+4$ is typically used as the return address.

First I thought that `following the JAL` means the instruction right after the `JAL` operation ($x+4$). After reading the section about `JAL` 

> JAL stores the address of the instruction following the jump

and the discussion in https://github.com/riscv/riscv-isa-manual/issues/356 about replacing `pc` with jump, I thought `jump` means the jump operation.
 
I propose the following:

> JAL stores the address of the instruction following the JAL instruction ({\tt pc}+4)

This wording could still be understood as: *after the JAL instruction has executed*. So another proposal is:

> JAL stores the address of the following instruction ({\tt pc}+4)

What do you think?

allenjbaum, 2022-08-30:
I'm usually good at interpreting even slightly ambiguous wording in a way
that was not intended, but not this case.
You understand it correctly. An even shorter wording might be
  "JAL stores the address of the instruction that follows it"
but your wording is even less subject to misinterpretation.

kasanovic, 2022-08-31:
I don't think it was that confusing before, but made an update.

closed 2022-08-31 by kasanovic

goekce, 2022-08-31:
Thanks for the comment and the update, now I find it less ambiguous (even without knowing that JAL should store the return address).

Note that the same phrase can be found in other sections. For consistency I would change them too:

https://github.com/riscv/riscv-isa-manual/search?q=%22following+the+jump%22

commit bf5a151c (2022-08-31): Clarified JAL behavior.
Closes #889.