riscv/riscv-isa-manual #890: Minor formatting and wording issues

goekce, 2022-08-31:
1. https://github.com/riscv/riscv-isa-manual/blob/bf5a151cda2a3e28a238f9c5f22575d38f7782b2/src/rv32.tex#L483

   vs

   https://github.com/riscv/riscv-isa-manual/blob/bf5a151cda2a3e28a238f9c5f22575d38f7782b2/src/rv32.tex#L605

   Is there a semantic difference between an instruction and operation?

   Not only these titles but many sentences use these words interchangeably.

2. https://github.com/riscv/riscv-isa-manual/blob/bf5a151cda2a3e28a238f9c5f22575d38f7782b2/src/rv32.tex#L631

    could use the same formatting as

    https://github.com/riscv/riscv-isa-manual/blob/bf5a151cda2a3e28a238f9c5f22575d38f7782b2/src/rv32.tex#L501

    (`ADD/SLT/SLTU` => `ADD/SLT[U]`)

Feel free to close this issue if they are too minor.

goekce, 2022-09-05:
Thanks @aswaterman. I'll leave the issue open, as point 1. was not addressed.

aswaterman, 2025-12-09:
Indeed, will fix.