riscv/riscv-isa-manual #891: Alignment requirements for hlv/hsv

dramforever, 2022-09-02:
Are the hypervisor load/store instructions guaranteed to not raise address misaligned exceptions in HS-mode when the address is misaligned?

As a special case, are the alignment requirements for `hlvx.wu` and `hlvx.hu` different? Are they affected by `IALIGN`?

The hypervisor chapter does not seem to specify. As of writing, OpenSBI does not emulate misaligned `hlv.*` etc:

https://github.com/riscv-software-src/opensbi/blob/5019fd124b4c46e1581129c5154fc2cdd3b777ed/lib/sbi/sbi_misaligned_ldst.c

aswaterman, 2022-09-02:
Same as for regular loads and stores: it's implementation-defined whether they can raise exceptions.  And same for the `hlvx` instructions (IALIGN doesn't matter).

Strictly speaking, this is not an omission, since the text describing the behavior of misaligned loads and stores is generic, i.e., not specific to the loads and stores defined in that chapter.

cc @jhauser-us to see if he wants to add an explicit note to this effect in the hypervisor extension chapter.

dramforever, 2022-09-03:
Thanks, I think this is a satisfactory response. I was just wondering if these were special and they're apparently not.

An explicit note may be helpful, but I personally don't think it's strictly necessary.

closed 2022-09-03 by dramforever